solvespace/solvespace #194: no DOF markers when analyzing assembly with DOF > 0

wpwrak, 2017-02-08:
It seems that Analyze > Show Degrees of Freedom never shows DOF indications (large cyan squares) when analyzing a group that contains (only) an assembly.

Some examples, all with master (e2efefeb) of the usual suspect, https://gitlab.com/anelok/mexp.git

- alt13/an-cover.slvs, group g003-oled13 (1 DOF)
- alt13/an-side.slvs, group g003-an-cover (3 DOF)
- alt13/an-top.slvs, group g007-an-side (2 DOF)

Evil-Spirit, 2017-02-10:
@wpwrak, an-cover.slvs is false dof reporting... i can't figure out why... but if you will remove all the groups but oled13, same orientation constrains only 2 dof instead of 3. but if you are create the empty file and apply same-orientation, it will constrain 3 dof as it should.

Evil-Spirit, 2017-02-10:
@wpwrak, @whitequark  now there are no visualization for normals (rotational dofs) and radius dofs. So, we have to draw them as something different than points...

whitequark, 2017-02-10:
@Evil-Spirit How about the same cyan line around the normal, or circle?

wpwrak, 2017-02-10:
@Evil-Spirit: I think I found it: if I delete the same-orientation constraint, switch to "Anywhere in 3d", and re-create it, I get DOF = 0. So the loose DOF was probably for real, caused by user error. Still, it should be reported.

Evil-Spirit, 2017-02-10:
The problem with assemblies is in fact what correspondent point doesn't exist. Imported group just generate some parameters for position and rotation quat which don't liked with any of the existent entities. So, there are nothing to visualize, we have to visualize it by different routine (per group) not inside point or any other entity.

whitequark, 2017-02-10:
@Evil-Spirit OK, then let's postpone it until we get rid of assembly groups (and just have assembly entity, maybe...)

Evil-Spirit, 2017-02-15:
@whitequark, updated here.

whitequark, 2017-02-17:
@Evil-Spirit This has a bug: Esc doesn't remove some highlighted DOFs.

Evil-Spirit, 2017-02-17:
@whitequark, hm, I will look into this.

whitequark, 2017-02-17:
@Evil-Spirit I fixed that, was a missing persistentDirty.

Evil-Spirit, 2017-02-17:
@whitequark, ok, thank you

whitequark, 2017-02-17:
@Evil-Spirit I've committed this. Can you look at test/request/circle/normal_dof ? For some reason in the test, there is no highlight of circle radius DOF. But it works in the GUI.

Evil-Spirit, 2017-02-17:
@whitequark, I have seen this already and confused too.

whitequark, 2017-02-17:
Thanks, fixed in master.

closed 2017-02-17 by whitequark