solvespace/solvespace #197: coexistence of "In Workplane" and "Anywhere in 3d"

wpwrak, 2017-02-10:
It it possible to make the "Sketch" menu show that both "In Workplane" and "Anywhere in 3d" are selected. Schroedinger would approve :-)

[image]

I've seen this on several occasions. Here is a somewhat circuitous way to reproduce it. I'm still looking for a simpler one.

solvespace master (f29d9d7a) plus small local changes
Let's use this example from https://gitlab.com/anelok/mexp.git master (3acc289)

- New
- assemble alt13/oled13.slvs
- select the point in the lower left corner of the panel
- select the origin and make both coincident (O)
- this produces an "unsolveable constraints" error (for a reason I haven't investigated yet)
- Undo
- select Sketch > In workplane (it has changed to "Anywhere in 3d")
- click away the complaint
- open Sketch > and be amazed

wpwrak, 2017-02-10:
Here is a shorter sequence:

- New
- assemble oled13.slvs (probably works with anything)
- Sketch -> Anywhere in 3d
- Sketch -> In Workplane
- click away the complaint
- Sketch > ... and there it is again

Note that, if we leave out the assembly step, we just get "In Workplane", as expected.

Evil-Spirit, 2017-02-17:
@wpwrak, It seems I can't reproduce it on Windows. And it's strange how the following code can cause this:
```
    RadioMenuByCmd(Command::FREE_IN_3D, !locked);
    RadioMenuByCmd(Command::SEL_WORKPLANE, locked);
```

Evil-Spirit, 2017-02-17:
@Schroedinger, can you explain it?

whitequark, 2017-02-17:
Seems like UB...

wpwrak, 2017-02-18:
UB = Ubuntu ? Yes, I'm on Ubuntu 16.10. I can collapse the quantum state (i.e., make it check only "Anywhere in 3d") by
https://0bin.net/paste/+c+6P42giR2hZiEl#AiwOfsiLHdPNCg2AOPR7Y3cM4g6xA2pY44gebEJmW7m
But I don't really know what I'm doing there, and if this may have nasty side-effects.

Symbian9, 2017-03-02:
> UB = Ubuntu ? Yes, I'm on Ubuntu 16.10

@wpwrak, What build of SolveSpace you use on Xenial now?

wpwrak, 2017-03-03:
@Symbian9: I just build from sources, with a number of fixes and improvement, either from @Evil-Spirit or self-made, that are not in mainline. Not sure about Xenial.

whitequark, 2019-05-23:
@wpwrak Are you still able to reproduce this? I've refactored a lot of the UI code in the meantime.

whitequark, 2019-07-31:
Have not been able to reproduce for two years, and no response in two months, closing.

whitequark, 2019-08-15:
I was able to reproduce it.
  
  1. Draw a rectangle
  2. Extrude it
  3. Hit `2`

whitequark, 2019-08-15:
OK, I figured it out. This is caused by rather obnoxious GTK behavior where set_active is not independent from causing a trigger event.

closed 2020-09-17 by phkahler

whitequark, 2020-09-17:
@phkahler I'm pretty sure this is not fixed--the underlying issue is the behavior of `set_active` in GTK which retriggers the action.

phkahler, 2020-09-17:
> @phkahler I'm pretty sure this is not fixed--the underlying issue is the behavior of `set_active` in GTK which retriggers the action.

What do you mean? On the surface this seems to fix it. Are you saying the condition may reappear?

whitequark, 2020-09-21:
> Are you saying the condition may reappear?

I don't know how to trigger it, but there is still a latent bug left in the GTK UI code, yes.