solvespace/solvespace #199: same orientation constrains two DOF when in workplane

wpwrak, 2017-02-10:
According to @whitequark same-orientation has no "In Workplane" variant. SolveSpace thinks it does:

Same setup as #197.

- New
- assemble oled13.slvs
- select #XY and Sketch > In Workplane
- notice that we have 6 DOF
- rotate up a little until a downward-pointing normal appears
- select that normal and the Z+ normal of the coordinate system
- same-orientation (X)
- we now have 4 DOF

[image]

whitequark, 2017-02-12:
@Evil-Spirit What's up with it? Reading the code I don't think the same-orientation constraint should be workplane-dependent, where am I wrong?

Evil-Spirit, 2017-02-13:
@whitequark, When we are trying to get NormalGetExprs which is using during builing equations for same orientation, it returns projected equations when in workplane. I don't know how we should do in this case. Probably, we should restrict using pure 3d constraints when operating in workplane,, but this is not good for backward compatibility

whitequark, 2017-02-13:
@jwesthues Is the same orientation constraint intended to produce different equations when in workplane? Or is that an accident?

jwesthues, 2017-02-13:
Not intentional, and I can't repro in 2.0. `NormalGetExprs` didn't used to return projected normals, and that wouldn't make a lot of sense. Are you sure the difference isn't the condition in [`AddEq()`](https://github.com/solvespace/solvespace/blame/e6ed36f739eaa29b827c9bc20d4434d5aa85ac59/src/constrainteq.cpp#L193), invoked from [here](https://github.com/solvespace/solvespace/blame/e6ed36f739eaa29b827c9bc20d4434d5aa85ac59/src/constrainteq.cpp#L614)?

Evil-Spirit, 2017-02-13:
@jwesthues, yes, you are right. This is because AddEq()

jwesthues, 2017-02-13:
@Evil-Spirit The geometrically obvious thing, constrain the projection of the point into the workplane to lie on the projection of the line into the workplane. That's what it did before.

Also, this has nothing to do with whether the entities are in 2d or 3d. This relates only to whether the constraint is projected into a workplane. You can sketch a line in workplane A, a point in workplane B, and constraint point-on-line, projected into workplane C.

Evil-Spirit, 2017-02-14:
@jwesthues, Then it is ok, because pt-on-line work using:

```
ExprVector EntityBase::PointGetExprsInWorkplane(hEntity wrkpl) const {
    if(wrkpl.v == Entity::FREE_IN_3D.v) {
        return PointGetExprs();
    }

    ExprVector r;
    PointGetExprsInWorkplane(wrkpl, &r.x, &r.y);
    r.z = Expr::From(0.0);
    return r;
}
```
So, we have projected version of line and points:
```
            EntityBase *ln = SK.GetEntity(entityA);
            EntityBase *a = SK.GetEntity(ln->point[0]);
            EntityBase *b = SK.GetEntity(ln->point[1]);
            EntityBase *p = SK.GetEntity(ptA);

            ExprVector ep = p->PointGetExprsInWorkplane(workplane);
            ExprVector ea = a->PointGetExprsInWorkplane(workplane);
            ExprVector eb = b->PointGetExprsInWorkplane(workplane);

            ExprVector ptOnLine = ea.Plus(eb.Minus(ea).ScaledBy(Expr::From(h.param(0))));
            ExprVector eq = ptOnLine.Minus(ep);

            AddEq(l, eq);
```

jwesthues, 2017-02-14:
Seems complicated. You can just use the same one equation as projected point-line-distance, with distance of zero.

Evil-Spirit, 2017-02-14:
@jwesthues, Then we will loose the direct parameter t and we can't do some ability to fix it (0.5=midpoint, any other cases also can be useful). This can be very useful when we are implement inequality (ranged constraints e.g. https://www.youtube.com/watch?v=u4dQ-eBAXx0).

jwesthues, 2017-02-14:
If you need t later, then you can calculate it from the point and line's endpoints. You need the extra parameter to write continuous equations for "parallel" or "point-on-line" in 3d, but not in 2d.

Evil-Spirit, 2017-02-14:
Now this is already introduced, so there is no any reasonable argument to remove it and then back again. Complicated equations is not problem. Equations can look like complicated and can work good because they produce perfect derivatives(but ugly equations), but also it can be simple and produce simple and fast derivatives but can cause serious issues when trying to converge (increase number of newton steps). I have at least half of year of deep learning, experimenting, optimizing solver and equations. If equations is complicated it's just problem of EvalJacobian and this is easy to optimize. It's hard to tell about what equation is better and what is poor just looking at formulas...

Evil-Spirit, 2017-02-14:
@jwesthues, I also have some theroy about "transformation of dof": when we are constraining some of the dof, we have to express it directly by defining the new parameter. So, every time we take off some dof, we should introduce new parameters which directly express the resulting dofs and link them by equations with involved old parameters.

whitequark, 2017-02-14:
> Now this is already introduced, so there is no any reasonable argument to remove it and then back again.

We haven't had a 3.0 release yet. We may break backwards compat still, for a really good reason.

Evil-Spirit, 2017-02-14:
@whitequark, We haven't break anything when equations was rewritten to use parameter, isn't it? For one more checkbox inside textwindow for pt-on-line we can allow showing value label and fixing param to specified value.

jwesthues, 2017-02-14:
This shouldn't affect backwards compatibility either way. It's a question of speed and stability.

Can you show an example where the extra parameter makes something work better in 2d? I'm open to the change, if there's a benefit that offsets the cost of the extra equation.

Evil-Spirit, 2017-02-16:
@jwesthues, there is no benefit probably. It just brings ability to work with t like we work with dimension (make constraint point-on with t = 0.1, t = 0.2, etc.). When we are introduce ranged constraints (which is just a lot of UI work rather than solver), we can do the next : https://www.youtube.com/watch?v=u4dQ-eBAXx0

whitequark, 2017-02-17:
Fixed in master.

closed 2017-02-17 by whitequark