solvespace/solvespace #202: why does lathe produce three DOF ?

wpwrak, 2017-02-12:
"New group rotating active sketch" aka "lathe" produces a group with three DOF. I tried to pull and twist, but nothing seems to be willing to move. So what are these three DOF ?

I asked @whitequark on #solvespace, and he didn't have an answer either.

whitequark, 2017-02-12:
@Evil-Spirit any ideas?

Evil-Spirit, 2017-02-13:
@whitequark revolution axis direction parameters is created when generating lathe group... I probably made this like it was made for extrusion group but now it is obvious what this is not the right solution because we can't drive some parameters of the previous group...

Evil-Spirit, 2017-02-13:
@whitequark, This is probably bug, so we have to remake this in correct way.

Evil-Spirit, 2017-02-13:
@whitequark, fixed here

wpwrak, 2017-02-15:
Works great here. This was the last step to achieving the 0 DOF goal for my design :) Thanks !

Evil-Spirit, 2017-02-15:
@wpwrak, Thank YOU, because your bugreports are the best and you provide a lot of useful information :) I hope you will keep going because this makes the project better!

whitequark, 2017-02-15:
I second what @Evil-Spirit says.

whitequark, 2017-02-17:
Fixed in master.

closed 2017-02-17 by whitequark