solvespace/solvespace #205: choice of active group when saving has puzzling effect on assemblies

wpwrak, 2017-02-16:
When saving a design when a group other than the last one is activated, items in the groups following it are not available in assemblies. I think this is not intended, and it's rather confusing.

Using solvespace/master (e6ed36) with various recent fixes and improvement from @Evil-Spirit and myself.

First, make a simple design with more than one group:

- File > New
- draw a line at roughly 45 degrees
- create a new group, with workplane parallel to default
- draw a rectangle through the endpoints of the line
- File > Save (as a.slvs)

Then assemble it:

- File > New
- assemble a.slvs
- the screen should look like this:
[image]

Now change the active group:

- File > Open a.slvs
- activate g002
- File > Save

Assemble again (or reload, etc.). The screen now looks like this:
[image]

Detecting the problem is quite hard since loading a.slvs (to look for incorrect settings) yields this perfectly normal-looking result:
[image]

whitequark, 2017-02-17:
No, this is expected. It works in exactly the same way as hiding groups then saving and linking the file. It will work this way until #80.

closed 2017-02-17 by whitequark

wpwrak, 2017-02-18:
Nice rationalizing ;-) But it doesn't work the same: if you hide groups through "show", etc., this information is preserved when you load the file, so you can make the connection between "show" and availability of items in an assembly fairly easily. But if you hide them through activation, this is hidden by SolveSpace activating the last group when loading the file.

So when the user looks at a file with that problem, they see a state that looks identical to one that does not have the problem. And it's hard to guess that activation (which one would think of as simply an act of selecting, much like, say, picking a sheet in a spreadsheet, with the non-availability of later groups being merely a side-effect) has this "magic" consequence.

E.g., it took me about a month to figure this out. Before I was able to connect the dots, my mental image was that assembling is "lossy", suspecting that this information loss somehow be linked to extrusion.

whitequark, 2017-02-19:
@wpwrak Regardless of whether you have a good argument I don't think this can be fixed before #80...