solvespace/solvespace #206: Feature Request: Exporting solid STEP

urmilparikh, 2017-03-04:
The STEP file exported by SolveSpace contains only bounding surface.
Please add an option to export solid model.

whitequark, 2017-04-08:
@Jonimoose can you please verify that the issue here is that STEP files exported by SolveSpace do not share edges between different bounding surfaces in the same solid?

Jonimoose, 2017-04-16:
I just imported an exported solvespace model into 3 different commercial CAD and CAM software packages and they all imported it as a "solid". 

The software I used were Autodesk PowerShape, Autodesk FeatureCAM, Solidworks, and Vero Software EdgeCAM.

I did notice that FreeCAD seems to import the model differently after I've exported it from PowerShape as a step than it would exported straight from solvespace but that is the only software I've found that doesn't import the model as a "solid."

diggit, 2017-04-16:
I have same problem with non solid STEP models in KiCAD which uses OCE for STEP model handling. (as well as FreeCAD)

urmilparikh, 2017-04-17:
Autodesk 123D Design imports it as hollow model.

[image]

Source and step files: [cube.zip](https://github.com/solvespace/solvespace/files/925447/cube.zip)

Jonimoose, 2017-04-17:
[10mmcube.zip](https://github.com/solvespace/solvespace/files/925840/10mmcube.zip)
I've attached a a 10mm cube model that has been exported from PowerShape which does not exhibit the issue in FreeCAD, other than being about half the number of lines the main difference I can see is on line 42 and 43 of the powershape export:
`#28=SHAPE_REPRESENTATION_RELATIONSHIP('','',#114,#30);
#29=SHAPE_REPRESENTATION_RELATIONSHIP('','',#114,#115);`

and further down on lines 146 and 147:
`#114=SHAPE_REPRESENTATION('master',(#116),#186);
#115=SHAPE_REPRESENTATION('',(#117,#118),#186);`

I don't see any equivalent lines in the solvespace export but those lines are also not referenced anywhere else so I'm not sure if that is what is doing it.

Jonimoose, 2017-04-22:
I have tracked down the issue to being what I and @whitequark initially thought, the step file output not reusing the EDGE_CURVE objects along shared edges. 

I modified the 10mmcube file in the zip I attached in the previous comment so that it only contains 12 edge_curve entities and then imported the files into the OpenCascade CAD Assistant and with the only change being combining the edge curves it comes in a single object instead of separate surfaces.
[10mmcube-modded3.zip](https://github.com/solvespace/solvespace/files/948761/10mmcube-modded3.zip)
You should be able to diff the .step file in that zip against the one I uploaded earlier to see the exact differences.

urmilparikh, 2017-04-28:
I confirm that `10mmcube-modded3.step` shared by @Jonimoose does show up as solid in 123D Design as well.

phkahler, 2020-10-23:
Since solvespace internally tracks which 2 surfaces any give trim curve joins, it shouldn't be too hard to export the curves only once and reuse the STEP identifier for the curve for both surfaces in the step file. That would also result in smaller STEP files and may help some programs realize that it is a closed shell.

ruevs, 2020-10-23:
Yeah but the STEP export code will have to become much more structured and smart/object oriented ;-) ... right now it is as straight forward as it gets - go through all surfaces, dump them.

phkahler, 2020-10-23:
Go through all curves and dump them while also creating a map going from internal ID to step file ID. Then dump surfaces, but dont do the curves along with them, just use the map to write their Step IDs.

I agree the step code is crude, but it works ;-)

phkahler, 2021-09-28:
@nanoant  Can you issue a pull request so we can have a look and try your changes?  I think it would be helpful to export a simple STEP file from a tool that does this correctly, as we really don't have a full understanding of the standard. If edges need to be reversed for one surface, it must happen where that edge (or curve) is referenced by a surface because we know (believe) the curve should only appear in the file once. What is the proper syntax for this in STEP? I don't know, so we need a correct example.

>> Altogether I believe outputting STEP that can be consumed as a solid by FreeCad or Fusion is pretty important if you want to share models and so some additional refinement, e.g. adding fillets.

Completely agree and BTW, I very much appreciate you trying to do this.

phkahler, 2021-09-28:
Searching for a simple cube I ran across this one with a hole in it:
https://isr.umd.edu/Labs/CIM/vm/xlator/step.html

Where entity IDs #9 and #27 both refer to #8 with the only difference being .F.  or  .T.  which is False or True. Searching on ORIENTED_EDGE brought me to this:
https://www.steptools.com/stds/stp_aim/html/t_oriented_edge.html

Which seems to indicate that the start and end of the edge are reversed based on the boolean value.

ruevs, 2021-09-28:
STEP is a... big... specification. Last year I superficially "dipped my toe in it" while implementing the STEP colors and transparency (https://github.com/solvespace/solvespace/pull/763, https://github.com/solvespace/solvespace/pull/767, https://github.com/solvespace/solvespace/issues/452). After that I [spent some time](https://github.com/solvespace/solvespace/issues/206#issuecomment-715521786) on this but unfortunately too little time.

Here is a dump of the references I found at the time, they may be useful:
1. STEP styling - used it to implement the color and transparency https://www.cax-if.org/documents/rec_prac_styling_org_v16.pdf
2. More where the above came from https://www.cax-if.org/cax/cax_recommPractice.php)
3. ISO 10303-21:2016 - the standard defining STEP (not free, search ;-) https://www.iso.org/standard/63141.html
4. The file format "schemas" https://www.cax-if.org/cax/cax_express.php (e.g. https://www.cax-if.org/documents/AP242ed2_HTML/AP242ed2.htm)
5. Some example models https://www.cax-if.org/cax/cax_stepLib.php

nanoant, 2021-09-28:
> Can you issue a pull request so we can have a look and try your changes?

@phkahler Well my code is available at https://github.com/nanoant/solvespace/tree/step/edge-reuse It has a bit of debugging junk in it so I am not sure if it is good candidate for PR at the moment, not to mention use of `std::map` and all approx stuff that is just quickly baked.

Anyway it seems to do the job for some not very complex shapes produced from some well defined boolean ops - FreeCad consumes exported STEP as a proper solid shape:

[image]

However if SS objects contain unions of some two adjacent arcs - we are still getting bad (non-solid) STEP files, e.g. this one:
[image]
(the edge marked red should not be there at all in FreeCad)

This may be the issue with my approx bezier matching/reusing, or maybe an issue with boolean ops in SS? I will look into that.

nanoant, 2021-09-28:
Oh.. okay, I think I know what is the problem. It looks like some issue with the solver, and the outcome depends on how you place constraints:

Doing this, leads to proper solid STEP file:
[image]

Doing this produces a file with right adjacent arcs edge broken:
[image]

Is this expected behavior?

nanoant, 2021-09-28:
Update. It looks like you need to compare bezier weights with some approximation when looking up the cache. I updated my commit with working code, please see / use this https://github.com/nanoant/solvespace/commit/2266fedbba846ec53e4f1ec292ee0621884a5dd3

phkahler, 2021-09-29:
@nanoant 
Some comments, I don't know what's best. I didn't expect to cache entire curves, but it turns out what I thought we could do isn't quite possible but may be with some help. Most of this is in the /srf folder, so see surface.h.

Some background:
A curve (SCurve) has a unique handle (h). It also has handles to the two surface it joins (surfA, surfB). They also have a flag (isExact) that tells if the curve is fully defined by the memeber (exact) which is an SBezier. Weather a curve is exact or not, it also has a piece-wise-linear (PWL) representation which is just a list of points (pts). All geometry that is created from a sketch via Extrude, Lathe, Revolve, or Helix will be composed of exact curves. Many intersections of surfaces can also be exact curves. When you get things like the intersection of 2 curved surfaces then the curve will not be exact, but will have only a PWL representation.

When a surface is exported to STEP, all the curves are exported. I thought we could just store a map of their handles, along with the STEP ID, so the second time they're written we'd just use the curve handle to look up the STEP ID.  The curves are also stored in an SShell object in a list, and the surfaces are also in there in a list so we could write the curves first by going over the list of curves. That's not quite possible.

One problem is the notion of direction. I don't recall where the direction flag is, but when we build an extrusion one curve handle gets flagged as reversed (I think) and the other not reversed.

The second (bigger) problem is the PWL (not exact) curves. When these are exported, each curve is converted to one or more cubic curves which will not have unique handles, and are apparently generated twice - once in each direction depending on the surface? This happens somewhere in FindOuterFacesFrom() and the functions it calls. I didn't realize SS does this entire step of creating an sblss at the time of export.

So the simple idea of creating a map of curve IDs is ruined by the one->many issue with non-exact curves, and also by the possible direction flipping going on in the sblss creation step. If those issues weren't there it should require less code than what you have. But by preserving info about which surfaces a curve belongs to and their directions it might take more than what you have, although there would be no need to try to match entire curves.

SolveSpace also has some functions that you implemented. You can compare Vectors using  a.Equals(b, tolerance) and maybe a few other things.

One more thought, I wonder if a PWL is converted to curves differently depending on the number of points and if it is reversed. That could make comparing curves impossible\, but I'm really not sure without studying more code.

Keep me posted on what you're up to and if I can help any more.

nanoant, 2021-09-29:
@phkahler Many thanks for such an exhaustive explanation. I have learned a lot.

Also I made some progress by realizing that `std::map` is a wrong container when using approximate comparisons (see https://github.com/nanoant/solvespace/commit/ba2be4a5ea6a3229b3f094f873cdde1cb42d3f86), since it requires strict weak ordering. I switched to `std::unordered_map` with curve id as a hash and now I can export most of my designs into FreeCad that picks them as solid object.

> All geometry that is created from a sketch via Extrude, Lathe, Revolve, or Helix will be composed of exact curves.

Studying `SSurface::MakeSectionEdgesInto` I can see some cases where even if the curve is exact, still not the whole (same) `SCurve.exact` is exported. Instead it is trimmed according to `ClosestPointTo` into two `ts` and `tf`.

This essentially happens when `ts>0` or `tf<1`. If `ts=0` and `tf=1` then it is should be exact. This seems to lead to numerical differences, which in the end require using approximate comparisons. I wonder if this code can be much such that we produce always the same numbers binary wise?

To be honest I don't fully understand yet the purpose of this mechanism.

> The second (bigger) problem is the PWL (not exact) curves. 

Yes that's the second case in `SSurface::MakeSectionEdgesInto`, although I must say that so far all of my designs had no PWL when I was dumping them. So far so good.

> One more thought, I wonder if a PWL is converted to curves differently depending on the number of points and if it is reversed. That could make comparing curves impossible, but I'm really not sure without studying more code.

Yeah that's true. But I guess it should be possible to ensure that PWLs are created always in same order, that should make produced Beziers also comparable.

> SolveSpace also has some functions that you implemented. You can compare Vectors using a.Equals(b, tolerance) and maybe a few other things.

Yes, now I realized that. Also I should probably use `LENGTH_EPS` rather than `0.0001f`. I will take a break at my work in 1-2 week, so I might have some more time to come with some useful PR.

phkahler, 2021-10-30:
[PWL_edges.zip](https://github.com/solvespace/solvespace/files/7447274/PWL_edges.zip
[image]

This file should have 8 surfaces. 4 for the main shape and 4 90 degree cylindrical surfaces for the hole. The top and botton edges of the hole should be PWL curves since they are the intersection of 2 curved surfaces. All other edges should be exact.

You could also change the extrusion to an intersection to output the "plug" for the hole, which should only have 6 surfaces with a combination of PWL curves and straight lines (the seams between the 90 degree sections).

phkahler, 2022-06-15:
@nanoant  Have you ever got a chance to do more on this? Do you have a branch here on github I can look at and maybe rebase to master?  If it's not too invasive I might want to use what you've got rather than waiting for a "perfect" solution that handles all cases.

ruevs, 2022-06-15:
@phkahler here https://github.com/nanoant/solvespace/tree/step/edge-reuse

raquo, 2023-03-25:
Sorry to butt in here... Is there any workaround for this at all, maybe some external script that could make a solvespace-exported step file solid? Couldn't find anything like that myself. Facing this issue with freecad as well.

ruevs, 2023-03-26:
Here is one possible approach that I have not tried personally. https://youtu.be/F3LJMeqUDrU

ilguido, 2025-05-19:
After some testing I can say that #1589 always export a valid, error-free STEP file, which is good for _most_ cases. The obvious problem with that happens when two different volumes have one common surface, that is when two surfaces from two different volumes coincide. That is a possibility especially with assemblies, e.g. the box assembly tutorial from the official site of Solvespace. That is not necessarily bad: applications that do not care about hidden surfaces (e.g. PrusaSlicer, a slicing engine for 3D printers) have no problem with that.

[image]

However that is a problem when one always needs actual volumes. So I made a new branch, with a new hack to mitigate this further problem. Since there is not, as far as I know, a simple way to tell one volume from the others in Solvespace, I used the colour of a surface as a way to distinguish different parts: the new algorithm does not compares just the coordinates of edges and surfaces, but also the colour of the surfaces (there is a switch to turn on and off this feature). Doing that, a user can apply colours to correctly export assemblies to the STEP file format. E.g. the box tutorial with its 10 volumes in Gmsh and Online 3D Viewer.

[image]

[image]

Testing branch with colour aware exporter: https://github.com/ilguido/solvespace/tree/testing

ruevs, 2025-05-20:
> Testing branch with colour aware exporter: https://github.com/ilguido/solvespace/tree/testing

@ilguido please make a new pull request based on this branch. I have a small change that I was about to propose in #1589 but it is your trunk. I will do it in your new PR.

When creating the pull request please "Allow edits and access to secrets by maintainers" so that I can rebase it on master and add my commit.