solvespace/solvespace #210: Write a tutorial that assumes no prior CAD knowledge whatsoever

whitequark, 2017-03-14:
Explain how to do the most basic steps:

1) draw 2D shape
1.5) add constraints to make it the size you want
2) make it 3D via rotation or extrusion

and don't use jargon etc.

tpimh, 2017-03-19:
I'd like to see this tutorial showing how to draw all sorts of 3D shapes. I now feel pretty confident with SolveSpace when drawing simple designs for 3D printing, but I still can't figure out how to create cones. Is it even possible?

[image]

whitequark, 2017-03-19:
@tpimh  Sure. Rotate a triangle.
[image]
[image]

tpimh, 2017-03-19:
Thanks! It's that simple.

Symbian9, 2017-03-27:
Also look on this
* https://github.com/solvespace/website/issues/1

romeojulietthotel, 2017-04-26:
This is a good one and it's basic.

:thumbsup:
https://youtu.be/1Em_XESnqUM