solvespace/solvespace #213: Text boxes are garbled during editing

jsmaniac, 2017-03-17:
### System information

SolveSpace version: (e.g. 3.0~911c67b2)

Operating system: NixOS 16.09 with custom up-to-date package from https://github.com/jsmaniac/nixpkgs/commit/d6f925d7c1ce29c2099e10dcbdb10628c1c63be1

### Expected behavior

What should have happened?

Text display shouldn't be garbled while editing it

### Actual behavior

Initially:
[image]

After clicking "change" (all fine):

[image]

After typing Hello (garbled display as soon as a key is pressed):
[image]

After pressing Enter (display is back to normal, editing did work):
[image]

### Additional information

Likely to be relevant library versions:

* fltk-1.3.3
* cairo-1.14.6
* mesa-12.0.4
* freetype-2.6.5
* fontconfig-2.11.1

whitequark, 2017-03-17:
> fltk-1.3.3

FLTK? SolveSpace uses GTK. Anyway, the problem you're observing is related to the theme you're using, and is the same as https://github.com/solvespace/solvespace/issues/184. Try installing a GTK theme.

jsmaniac, 2017-03-17:
I had updated an existing recipe for my OS, and solvespace 2 used FLTK. I should remove it from the dependencies.

Thanks for the pointer (I'm using Enlightenment 0.16, and it's not the first time I've seen missing stuff due to the lack of a GTK theme).

closed 2017-03-17 by jsmaniac

whitequark, 2017-03-17:
> I had updated an existing recipe for my OS, and solvespace 2 used FLTK. I should remove it from the dependencies.

Actually, no SolveSpace release used FLTK. There was a brief time when the open-sourced code had an FLTK implementation, but it was essentially unusable.

whitequark, 2017-03-19:
Incidentally, this is now fixed properly in master.

jsmaniac, 2017-03-20:
Sorry about the delay, I tested the latest (827f54f) commit on master, and the fix works perfectly. Thanks!