solvespace/solvespace #214: Memory leak

jsmaniac, 2017-03-17:
### System information

SolveSpace version: (e.g. 3.0~911c67b2)

Operating system: NixOS 16.09 with custom up-to-date package from https://github.com/jsmaniac/nixpkgs/commit/d6f925d7c1ce29c2099e10dcbdb10628c1c63be1

### Expected behavior

What should have happened?

Shouldn't crash due to OOM.

### Actual behavior

Memory leak / crashes with Out Of Memory

### Additional information

Steps to reproduce:

1. Start solvespace from terminal
2. Press tab twice to show up property browser (it is checked but doesn't show up in my machine by default, might be WM-related, as I use an old Enlightenment 0.16)
3. Scroll down and click "Create user style"
4. Click on that style
5. Scroll up and click [change] next to the colour
6. The memory usage of solvespace increases by about 10MB/s
7. As soon as I click outside of the colour selection dialog to dismiss it, the memory usage stops growing

This could be related to https://github.com/solvespace/solvespace/issues/213, but the memory usage does not increase noticeably while editing other text fields, it only increases when the colour selection box is visible.

whitequark, 2017-03-17:
> Scroll down and click "Create user style"

Do you mean "create a new custom style"? In any case, I cannot reproduce this, so you'll have to figure out what causes the allocations yourself. Try running solvespace under `valgrind`.

jsmaniac, 2017-03-17:
Yes, sorry, I meant "create a new custom style". I'll try with valgrind later (hopefully this weekend).

jsmaniac, 2017-03-17:
An initial run of valgrind spits out a plethora of warnings etc.

This stacktrace looks promissing:

whitequark, 2017-03-17:
A lot of warnings comes from GTK. It seems to be rather sloppily written, and it's nearly impossible to produce a decent suppression list (I tried).

You didn't post a stacktrace.

jsmaniac, 2017-03-17:
An initial run of valgrind spits out a plethora of warnings etc.

This stacktrace looks promissing:
```
==16793==    by 0x48AF05: SolveSpace::OpenGl2Renderer::SelectTexture(std::shared_ptr<SolveSpace::Pixmap const>) (in /nix/store/9f1k6vmslifhisv2i8diqgklhdjd1316-solvespace-3.0/bin/solvespace)
==16793==    by 0x48B83A: SolveSpace::OpenGl2Renderer::SelectFill(SolveSpace::Canvas::hFill) (in /nix/store/9f1k6vmslifhisv2i8diqgklhdjd1316-solvespace-3.0/bin/solvespace)
==16793==    by 0x48BC02: SolveSpace::OpenGl2Renderer::FlushFrame() (in /nix/store/9f1k6vmslifhisv2i8diqgklhdjd1316-solvespace-3.0/bin/solvespace)
==16793==    by 0x54A8D5: SolveSpace::TextWindow::Paint() (in /nix/store/9f1k6vmslifhisv2i8diqgklhdjd1316-solvespace-3.0/bin/solvespace)
==16793==    by 0x47E583: SolveSpace::TextWidget::on_render(Glib::RefPtr<Gdk::GLContext> const&) (in /nix/store/9f1k6vmslifhisv2i8diqgklhdjd1316-solvespace-3.0/bin/solvespace)
==16793==    by 0x55F7FB3: Gtk::GLArea_Class::render_callback(_GtkGLArea*, _GdkGLContext*) (in /nix/store/pvi2kjd5kj771cic6c3ca0zk0ws0sb4p-gtkmm-3.20.0/lib/libgtkmm-3.0.so.1.1.0)
==16793==    by 0x111F50FF: ffi_call_unix64 (in /nix/store/lh7xniswq51mz9gwxwv081nqll1agdg5-libffi-3.2.1/lib/libffi.so.6.0.4)
==16793==    by 0x111F4B6A: ffi_call (in /nix/store/lh7xniswq51mz9gwxwv081nqll1agdg5-libffi-3.2.1/lib/libffi.so.6.0.4)
==16793==    by 0x71287D4: g_cclosure_marshal_generic_va (in /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0.4800.2)
==16793==    by 0x7127CB3: _g_closure_invoke_va (in /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0.4800.2)
==16793==    by 0x7141F67: g_signal_emit_valist (in /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0.4800.2)
```

I'm having trouble navigating the valgrind log, though, and I can't find an AlleyOop nor Valkyrie package for my distro :-( . I'll dig more into this later.

whitequark, 2017-03-17:
> ==16793== 4,841,661 (56 direct, 4,841,605 indirect) bytes in 1 blocks are definitely lost in loss record 13,472 of 13,472

This doesn't seem concerning. This is just the font texture, it's quite large but it is only ever allocated once.

jsmaniac, 2017-03-17:
Sorry, I'm rarely ever coding in C/C++, so I'm not used to valgrind. I tried with `--tool=massif`, in the graph below the first ramping up is when I started solvespace, then I left it rest for a short while (the flat section at the beginning), then I opened the colour picker (the second, more steady upward slope). I took screenshots of the green, turquoise and purple blocks on the last time frame.

I uploaded the massif report file [massif.out.30666.zip](https://github.com/solvespace/solvespace/files/851389/massif.out.30666.zip), but I can't make much sense of it (it seems that the root of the stack traces is gtk_widget_draw_internal for the last two cases, which is rather nonspecific.

[image]
[image]

whitequark, 2017-03-18:
Okay, so this is a texture-related leak. Can you apply this patch and show the output in the console?
```patch
diff --git a/src/render/rendergl2.cpp b/src/render/rendergl2.cpp
index 5830199f..af710904 100644
--- a/src/render/rendergl2.cpp
+++ b/src/render/rendergl2.cpp
@@ -18,6 +18,7 @@ public:
         if(it == items.end()) {
             GLuint id;
             glGenTextures(1, &id);
+            dbp("creating texture %u", id);
             items[ptr] = id;
             *result = id;
             return false;
```

Evil-Spirit, 2017-03-19:
@whitequark, this probably should fix it.

jsmaniac, 2017-03-20:
Indeed, with whitequark's patch, I see a lot (hundreds every second) of `creating texture xxxx` messages as soon as the colour selection box is opened (and none printed in other situations.

```
> solvespace                                                                                                                                                                                                            phc  ~/git/nixpkgs

** (solvespace:23909): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
Missing (absent) translation for '&Image'
Missing (absent) translation for 'Show &Center of Mass'
SolveSpace!

creating texture 10
creating texture 11
creating texture 12
creating texture 13
creating texture 14
creating texture 15
creating texture 16
creating texture 17
creating texture 18
creating texture 19
creating texture 20
creating texture 21
creating texture 10
creating texture 11
creating texture 12
creating texture 13
creating texture 14
creating texture 15
…
```

I tried @Evil-Spirit's branch, and it does fix the memory leak. Thanks!

whitequark, 2017-03-20:
@Evil-Spirit I don't understand how the memory leak actually works. If you look at the log, you'll see that we're overwriting the same textures over and over again. Why does that increase memory usage?

Also, I got this report from IRC:
```
11:54 < lexszero> has anybody observed memory leak problems with solvespace? just ran out of 4G of ram and 10G of swap, which was instantly freed with
                  killing solvespace (that i didn't use at the moment and it was silently idling in background for few hours).
11:55 < lexszero> the strange thing is the solvespace process RSS was not that big at all (neither any other process in the system), so i'm trying to figure
                  out WTF.
```
This looks as if the *driver itself* has a memory leak. Are graphics drivers really that shitty?

Evil-Spirit, 2017-03-21:
@whitequark,
We are allocting new pixmap every time we draw color dialog, but this is deleted by shared_ptr.  But also we create new opengl texture and never free it.

whitequark, 2017-03-22:
Fixed in master.

closed 2017-03-22 by whitequark