solvespace/solvespace #215: File→Save and File→Save As… crash SolveSpace

jsmaniac, 2017-03-17:
### System information

SolveSpace version: (e.g. 3.0~911c67b2)

Operating system: NixOS 16.09 with custom up-to-date package from https://github.com/jsmaniac/nixpkgs/commit/d6f925d7c1ce29c2099e10dcbdb10628c1c63be1

### Expected behavior

What should have happened?

A File Save dialog box should appear when pressing Ctrl+S or File→Save or File→Save As…

### Actual behavior

SolveSpace crashes. Here's what I get in the console:

```
> solvespace                                                                                                                                                                                                            phc  ~/git/nixpkgs

** (solvespace:3974): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
Missing (absent) translation for '&Image'
Missing (absent) translation for 'Show &Center of Mass'
SolveSpace!

(solvespace:3974): GLib-GIO-ERROR **: No GSettings schemas are installed on the system
zsh: trace trap  solvespace
```

### Additional information

whitequark, 2017-03-17:
This seems like a NixOS compatibility issue. Exactly where does it crash? See https://github.com/solvespace/solvespace/blob/master/CONTRIBUTING.md#debugging-a-custom-build for instructions on how to gather a backtrace.

whitequark, 2017-03-17:
Actually, you've mentioned that the property browser window did not instantly appear. What's happening is I'm calling `show_all()` right away on that window. If GTK does the same, and then expects the window to be realized, and it's not, then it will probably die dereferencing a null pointer or something like that.

jsmaniac, 2017-03-17:
It turns out the browser window was just hiding behind the main window (it seems like the "foreground window" property is not taken into account, perhaps because the main window appears *after* the browser window. Even if I move the main window to the side, and bring the browser window into focus, it goes to the back as soon as focus goes to the main window. After pressing tab twice, the browser window correctly stays on top even when the main window has the focus). So it must be something else.

whitequark, 2017-03-17:
> Even if I move the main window to the side, and bring the browser window into focus, it goes to the back as soon as focus goes to the main window. After pressing tab twice, the browser window correctly stays on top even when the main window has the focus).

Does the following patch fix things for you right after startup?

```patch
diff --git a/src/platform/gtkmain.cpp b/src/platform/gtkmain.cpp
index a1e5bca9..01d46f40 100644
--- a/src/platform/gtkmain.cpp
+++ b/src/platform/gtkmain.cpp
@@ -1424,7 +1424,6 @@ int main(int argc, char** argv) {
 
     TW.reset(new TextWindowGtk);
     GW.reset(new GraphicsWindowGtk);
-    TW->set_transient_for(*GW);
     GW->set_icon(icon_gdk);
     TW->set_icon(icon_gdk);
 
```

jsmaniac, 2017-03-17:
No, that patch didn't change anything (I tried adding `TW->set_transient_for(*GW);` after the `GW->show_all();`, still not good). I wouldn't worry about the initial display of the browser window, though, as it is most likely a problem with my window manager, with a simple workaround (press tab twice).

Here's the backtrace for the Save/Save As… crash:
```
> gdb $(readlink -f $(which solvespace))                                                                                                                                                                                                                                                                           phc  ~
GNU gdb (GDB) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /nix/store/hk1l2zr663xf6ppm1vb6hhliz5la7jii-solvespace-3.0/bin/solvespace...(no debugging symbols found)...done.
(gdb) run
Starting program: /nix/store/hk1l2zr663xf6ppm1vb6hhliz5la7jii-solvespace-3.0/bin/solvespace
warning: File "/nix/store/djd2r4cnbcx4vfbj1qalg85vjbcslwxv-gcc-5.4.0-lib/lib/libstdc++.so.6.0.21-gdb.py" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
	add-auto-load-safe-path /nix/store/djd2r4cnbcx4vfbj1qalg85vjbcslwxv-gcc-5.4.0-lib/lib/libstdc++.so.6.0.21-gdb.py
line to your configuration file "/home/georges/.gdbinit".
To completely disable this security protection add
	set auto-load safe-path /
line to your configuration file "/home/georges/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
	info "(gdb)Auto-loading safe path"
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/nix/store/q3wx1gab2ysnk5nyvyyg56ana2v4r2ar-glibc-2.24/lib/libthread_db.so.1".

** (solvespace:31886): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
[New Thread 0x7fffe14ab700 (LWP 31905)]
[New Thread 0x7fffe0caa700 (LWP 31906)]
Missing (absent) translation for '&Image'
Missing (absent) translation for 'Show &Center of Mass'
SolveSpace!

[New Thread 0x7fffd3b37700 (LWP 31927)]
[New Thread 0x7fffd3336700 (LWP 31928)]
[New Thread 0x7fffd2b35700 (LWP 31929)]
[New Thread 0x7fffd2334700 (LWP 31930)]
[New Thread 0x7fffd1b33700 (LWP 31931)]
[New Thread 0x7fffd1332700 (LWP 31932)]
[New Thread 0x7fffd0b31700 (LWP 31933)]
[New Thread 0x7fffd0330700 (LWP 31934)]
[New Thread 0x7fffcfb2f700 (LWP 31935)]
[New Thread 0x7fffcf32e700 (LWP 31936)]

(solvespace:31886): GLib-GIO-ERROR **: No GSettings schemas are installed on the system

Thread 1 "solvespace" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff55e7f7b in g_logv () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libglib-2.0.so.0
(gdb) backtrace
#0  0x00007ffff55e7f7b in g_logv () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libglib-2.0.so.0
#1  0x00007ffff55e80ef in g_log () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libglib-2.0.so.0
#2  0x00007fffefd51e22 in g_settings_set_property () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgio-2.0.so.0
#3  0x00007ffff58bc17b in g_object_new_internal () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0
#4  0x00007ffff58bda53 in g_object_new_valist () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0
#5  0x00007ffff58bdfa1 in g_object_new () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0
#6  0x00007ffff66334dc in _gtk_file_chooser_get_settings_for_widget () from /nix/store/y7z5933vx4xwbp34n82jlxbch423lv2s-gtk+3-3.20.9/lib/libgtk-3.so.0
#7  0x00007ffff6636ae0 in gtk_file_chooser_widget_get_default_size () from /nix/store/y7z5933vx4xwbp34n82jlxbch423lv2s-gtk+3-3.20.9/lib/libgtk-3.so.0
#8  0x00007ffff6630374 in file_chooser_widget_default_size_changed () from /nix/store/y7z5933vx4xwbp34n82jlxbch423lv2s-gtk+3-3.20.9/lib/libgtk-3.so.0
#9  0x00007ffff58b6a85 in g_closure_invoke () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0
#10 0x00007ffff58c8a71 in signal_emit_unlocked_R () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0
#11 0x00007ffff58d180c in g_signal_emit_valist () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0
[... 24 lines cut ...]
#36 0x00007ffff58d12a9 in g_signal_emit_valist () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0
#37 0x00007ffff58d1b3f in g_signal_emit () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libgobject-2.0.so.0
[... cut ...]
#54 0x00007ffff55e18f0 in g_main_context_iterate.isra () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libglib-2.0.so.0
#55 0x00007ffff55e1c12 in g_main_loop_run () from /nix/store/624597s5chn14ifvcqyw7qw2yfkgcp0f-glib-2.48.2/lib/libglib-2.0.so.0
#56 0x00007ffff669b305 in gtk_main () from /nix/store/y7z5933vx4xwbp34n82jlxbch423lv2s-gtk+3-3.20.9/lib/libgtk-3.so.0
#57 0x00007ffff7553e91 in Gtk::Main::run(Gtk::Window&) () from /nix/store/pvi2kjd5kj771cic6c3ca0zk0ws0sb4p-gtkmm-3.20.0/lib/libgtkmm-3.0.so.1
#58 0x000000000046e13b in main ()
(gdb)
```

whitequark, 2017-03-18:
Ah, yes, I see. It fails an assertion in GTK, specifically this one:
```
(solvespace:31886): GLib-GIO-ERROR **: No GSettings schemas are installed on the system
```
This is a misconfiguration of the desktop environment on your system that I can do nothing about (and have no particular insight into). Try the suggestion in https://stackoverflow.com/questions/37510261/ubuntu-meld-no-gsettings-schemas-are-installed-on-the-system#37578829.

closed 2017-03-18 by whitequark