solvespace/solvespace #217: Add external libs in sources archive.

SL-RU, 2017-03-19:
### System information

SolveSpace version: 2.3

Operating system: Arch linux

### Expected behavior

Archive with sources must be absolutely complete. Anyone must have ability to build it even without network. But your release archives doesn't  contains external libs. That makes impossible to build it on offline machine.

### Actual behavior

All necessary sources must be  included in the archive.

whitequark, 2017-03-19:
The source archive is built by GitHub, and GitHub doesn't include submodules. Unfortunately there isn't anything I can do about it (even hide the links).

whitequark, 2017-03-19:
Also, the README explicitly instructs to use git for checking out sources.

closed 2017-03-19 by whitequark

SL-RU, 2017-03-19:
Thanks for the answer. I understand. But maybe than you will publish those archives on your website externally? Because for someone it can be a problem.

whitequark, 2017-03-19:
Is using git a problem? If you're using Linux, the git checkout is also far smaller, since most external libraries (i.e. except libdxfrw) are only needed on Windows.

SL-RU, 2017-03-19:
No, not git. Building solvespace without network is a problem) Usually when archive with release sources are published, that means anyone can download it and compile it without network and etc.

But anyway.