solvespace/solvespace #218: Unable to build

SL-RU, 2017-03-19:
### System information

SolveSpace version: 2.3

Operating system: Arch

### Expected behavior

I've cloned git repo, installed all dependencies and init submodules. But I've error.

[Here is stderr output](https://github.com/solvespace/solvespace/files/853653/log.txt)

### Additional information

`yaourt -S solvespace and` and `yaourt -S solvespace-git` are finishing with the same result

SL-RU, 2017-03-19:
I've also tried to checkout 2.x branch. Result was the same

whitequark, 2017-03-20:
Why is `ui.h` included from gcc's `limits.h`?
```
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/include-fixed/limits.h:168:0,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/include-fixed/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/include-fixed/limits.h:34,
                 from /home/lyra/tmp/solvespace/solvespace/src/solvespace.h:18,
                 from /home/lyra/tmp/solvespace/solvespace/src/util.cpp:7:
```
I'd like to see the contents of `/usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/include-fixed/limits.h` on your system.

SL-RU, 2017-03-20:
Here it is:

```C
[lyra@lyrathink ~]$ cat /usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/include-fixed/limits.h
/* Copyright (C) 1992-2016 Free Software Foundation, Inc.

This file is part of GCC.

GCC is free software; you can redistribute it and/or modify it under
the terms of the GNU General Public License as published by the Free
Software Foundation; either version 3, or (at your option) any later
version.

GCC is distributed in the hope that it will be useful, but WITHOUT ANY
WARRANTY; without even the implied warranty of MERCHANTABILITY or
FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License
for more details.

Under Section 7 of GPL version 3, you are granted additional
permissions described in the GCC Runtime Library Exception, version
3.1, as published by the Free Software Foundation.

You should have received a copy of the GNU General Public License and
a copy of the GCC Runtime Library Exception along with this program;
see the files COPYING3 and COPYING.RUNTIME respectively.  If not, see
<http://www.gnu.org/licenses/>.  */

/* This administrivia gets added to the beginning of limits.h
   if the system has its own version of limits.h.  */

/* We use _GCC_LIMITS_H_ because we want this not to match
   any macros that the system's limits.h uses for its own purposes.  */
#ifndef _GCC_LIMITS_H_  /* Terminated in limity.h.  */
#define _GCC_LIMITS_H_

#ifndef _LIBC_LIMITS_H_
/* Use "..." so that we find syslimits.h only in this same directory.  */
#include "syslimits.h"
#endif
/* Copyright (C) 1991-2016 Free Software Foundation, Inc.

This file is part of GCC.

GCC is free software; you can redistribute it and/or modify it under
the terms of the GNU General Public License as published by the Free
Software Foundation; either version 3, or (at your option) any later
version.

GCC is distributed in the hope that it will be useful, but WITHOUT ANY
WARRANTY; without even the implied warranty of MERCHANTABILITY or
FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License
for more details.

Under Section 7 of GPL version 3, you are granted additional
permissions described in the GCC Runtime Library Exception, version
3.1, as published by the Free Software Foundation.

You should have received a copy of the GNU General Public License and
a copy of the GCC Runtime Library Exception along with this program;
see the files COPYING3 and COPYING.RUNTIME respectively.  If not, see
<http://www.gnu.org/licenses/>.  */

#ifndef _LIMITS_H___
#define _LIMITS_H___

/* Number of bits in a `char'.  */
#undef CHAR_BIT
#define CHAR_BIT __CHAR_BIT__

/* Maximum length of a multibyte character.  */
#ifndef MB_LEN_MAX
#define MB_LEN_MAX 1
#endif

/* Minimum and maximum values a `signed char' can hold.  */
#undef SCHAR_MIN
#define SCHAR_MIN (-SCHAR_MAX - 1)
#undef SCHAR_MAX
#define SCHAR_MAX __SCHAR_MAX__

/* Maximum value an `unsigned char' can hold.  (Minimum is 0).  */
#undef UCHAR_MAX
#if __SCHAR_MAX__ == __INT_MAX__
# define UCHAR_MAX (SCHAR_MAX * 2U + 1U)
#else
# define UCHAR_MAX (SCHAR_MAX * 2 + 1)
#endif

/* Minimum and maximum values a `char' can hold.  */
#ifdef __CHAR_UNSIGNED__
# undef CHAR_MIN
# if __SCHAR_MAX__ == __INT_MAX__
#  define CHAR_MIN 0U
# else
#  define CHAR_MIN 0
# endif
# undef CHAR_MAX
# define CHAR_MAX UCHAR_MAX
#else
# undef CHAR_MIN
# define CHAR_MIN SCHAR_MIN
# undef CHAR_MAX
# define CHAR_MAX SCHAR_MAX
#endif

/* Minimum and maximum values a `signed short int' can hold.  */
#undef SHRT_MIN
#define SHRT_MIN (-SHRT_MAX - 1)
#undef SHRT_MAX
#define SHRT_MAX __SHRT_MAX__

/* Maximum value an `unsigned short int' can hold.  (Minimum is 0).  */
#undef USHRT_MAX
#if __SHRT_MAX__ == __INT_MAX__
# define USHRT_MAX (SHRT_MAX * 2U + 1U)
#else
# define USHRT_MAX (SHRT_MAX * 2 + 1)
#endif

/* Minimum and maximum values a `signed int' can hold.  */
#undef INT_MIN
#define INT_MIN (-INT_MAX - 1)
#undef INT_MAX
#define INT_MAX __INT_MAX__

/* Maximum value an `unsigned int' can hold.  (Minimum is 0).  */
#undef UINT_MAX
#define UINT_MAX (INT_MAX * 2U + 1U)

/* Minimum and maximum values a `signed long int' can hold.
   (Same as `int').  */
#undef LONG_MIN
#define LONG_MIN (-LONG_MAX - 1L)
#undef LONG_MAX
#define LONG_MAX __LONG_MAX__

/* Maximum value an `unsigned long int' can hold.  (Minimum is 0).  */
#undef ULONG_MAX
#define ULONG_MAX (LONG_MAX * 2UL + 1UL)

#if defined (__STDC_VERSION__) && __STDC_VERSION__ >= 199901L
/* Minimum and maximum values a `signed long long int' can hold.  */
# undef LLONG_MIN
# define LLONG_MIN (-LLONG_MAX - 1LL)
# undef LLONG_MAX
# define LLONG_MAX __LONG_LONG_MAX__

/* Maximum value an `unsigned long long int' can hold.  (Minimum is 0).  */
# undef ULLONG_MAX
# define ULLONG_MAX (LLONG_MAX * 2ULL + 1ULL)
#endif

#if defined (__GNU_LIBRARY__) ? defined (__USE_GNU) : !defined (__STRICT_ANSI__)
/* Minimum and maximum values a `signed long long int' can hold.  */
# undef LONG_LONG_MIN
# define LONG_LONG_MIN (-LONG_LONG_MAX - 1LL)
# undef LONG_LONG_MAX
# define LONG_LONG_MAX __LONG_LONG_MAX__

/* Maximum value an `unsigned long long int' can hold.  (Minimum is 0).  */
# undef ULONG_LONG_MAX
# define ULONG_LONG_MAX (LONG_LONG_MAX * 2ULL + 1ULL)
#endif

#endif /* _LIMITS_H___ */
/* This administrivia gets added to the end of limits.h
   if the system has its own version of limits.h.  */

#else /* not _GCC_LIMITS_H_ */

#ifdef _GCC_NEXT_LIMITS_H
#include_next <limits.h>		/* recurse down to the real one */
#endif

#endif /* not _GCC_LIMITS_H_ */
```

whitequark, 2017-03-20:
Based on your log, `#include_next <limits.h>` causes gcc to include SolveSpace's `ui.h` instead of the system header (which may not even exist). There is no conceivable reason whatsoever for this behavior, and it is almost certainly some issue with your installation. (I believe SolveSpace is in AUR and other people have never encountered this).

I suggest creating a fresh Arch chroot and reproducing the problem there. If it persists, send me the tarball of that chroot and I'll look into it.

jonas-schievink, 2017-03-20:
I can reproduce on Arch with both GCC and Clang.

Clang reports:

```
../../src/ui.h:279:9: error: expected identifier
        CHAR_WIDTH     = 9,
        ^
/usr/include/limits.h:147:22: note: expanded from macro 'CHAR_WIDTH'
#  define CHAR_WIDTH 8
                     ^
```

The GCC error looks pretty much the same.

whitequark, 2017-03-20:
Oh *facedesk*

whitequark, 2017-03-20:
@jonas-schievink Can you replace all instances of `CHAR_WIDTH` with, say, `CHAR_WIDTH_` and see if that helps?

jonas-schievink, 2017-03-20:
@whitequark Yep, that makes it work.

FYI, I've been able to build Solvespace in the past, but even older commits fail to compile with the same error now, so this must have been caused or exposed by some updated package in Arch.

whitequark, 2017-03-20:
Yeah, I think it's an update in glibc.

diggit, 2017-03-31:
IMO there are several options how to avoid this collision:
- use library prefix (quite common) like: UI_CHAR_WIDTH
  - many projects have such requirement in their source code guidelines to prevent exactly these scenarios
  - if used, other macros should adopt prefix too to preserve code readability
- use different abbreviation or just whole word like: CHARACTER_WIDTH (workaround)
- use #ifdef... #undef... after limits.h include directive (pretty bad workaround)

Fixing is quite easy, but the way how to fix it, is must be chosen.

whitequark, 2017-03-31:
I'm quite aware of the ways to fix it.

diggit, 2017-03-31:
May I offer PR with your preferred type of fix? This is simple bug. So why wait?

whitequark, 2017-03-31:
Because I'm busy. No PRs please, since we still don't have a CLA, and I couldn't merge it anyway.

whitequark, 2017-04-06:
Wow, this is a standards disaster. Anyway, fixed in 7eb6574f.

whitequark, 2017-04-06:
Also in the 2.x branch in 7c2417a.

closed 2017-04-06 by whitequark

SL-RU, 2017-04-06:
Thanks a lot!