solvespace/solvespace #219: Ninja warnings:: "multiple rules generate *.po"

tpimh, 2017-03-20:
This is a very minor issue (but it may be annoying).

CMake generated `ninja.build` contains multiple rules for each `.po` file and complains about it:

    ninja: warning: multiple rules generate res/locales/en_US.po. builds involving this target will not be correct; continuing anyway
    ninja: warning: multiple rules generate res/locales/uk_UA.po. builds involving this target will not be correct; continuing anyway

Generated build.ninja:

    …
    
    #############################################
    # Custom command for res/locales/en_US.po

    build res/locales/en_US.po: CUSTOM_COMMAND ../res/locales/en_US.po
      COMMAND = cd /path/to/solvespace/build/res && /usr/bin/cmake -E make_directory /path/to/solvespace/build/res/locales && /usr/bin/cmake -E copy /path/to/solvespace/res/locales/en_US.po /path/to/solvespace/build/res/locales/en_US.po
      DESC = Copying resource locales/en_US.po
      restat = 1

    #############################################
    # Custom command for res/locales/uk_UA.po

    build res/locales/uk_UA.po: CUSTOM_COMMAND ../res/locales/uk_UA.po
      COMMAND = cd /path/to/solvespace/build/res && /usr/bin/cmake -E make_directory /path/to/solvespace/build/res/locales && /usr/bin/cmake -E copy /path/to/solvespace/res/locales/uk_UA.po /path/to/solvespace/build/res/locales/uk_UA.po
      DESC = Copying resource locales/uk_UA.po
      restat = 1
    
    …
    
    #############################################
    # Custom command for res/locales/en_US.po

    build res/locales/en_US.po: CUSTOM_COMMAND res/messages.pot
      COMMAND = cd /path/to/solvespace/build/src && /usr/bin/msginit --locale=en_US --no-translator --output=/path/to/solvespace/build/src/../res/messages.po --input=/path/to/solvespace/build/src/../res/messages.pot && /usr/bin/msgmerge --force-po --no-fuzzy-matching --output=/path/to/solvespace/build/src/../res/locales/en_US.po /path/to/solvespace/src/../res/locales/en_US.po /path/to/solvespace/build/src/../res/messages.po && /usr/bin/cmake -E copy_if_different /path/to/solvespace/build/src/../res/locales/en_US.po /path/to/solvespace/src/../res/locales/en_US.po
      DESC = Updating en_US translations
      restat = 1

    #############################################
    # Custom command for res/locales/uk_UA.po

    build res/locales/uk_UA.po: CUSTOM_COMMAND res/messages.pot
      COMMAND = cd /path/to/solvespace/build/src && /usr/bin/msgmerge --no-fuzzy-matching --output=/path/to/solvespace/build/src/../res/locales/uk_UA.po /path/to/solvespace/src/../res/locales/uk_UA.po /path/to/solvespace/build/src/../res/messages.pot && /usr/bin/cmake -E copy_if_different /path/to/solvespace/build/src/../res/locales/uk_UA.po /path/to/solvespace/src/../res/locales/uk_UA.po
      DESC = Updating uk_UA translations
      restat = 1
    
    …

### System information

- SolveSpace version 3.0~827f54f2
- Operating system: GNU/Linux x86_64
- CMake 3.5.1
- Ninja 1.5.1

whitequark, 2017-03-20:
No, it's fair, and can likely cause hard to debug race conditions.

whitequark, 2017-04-08:
Fixed in master.

closed 2017-04-08 by whitequark