solvespace/solvespace #226: Imprecise unsolvable constraint error

whitequark, 2017-04-05:
See the following file: https://drive.google.com/file/d/0B15RHGNY8nBpSU1xNjJHbmpyNnM/view, created in 2.1 originally.
There are lots of unsolvable constraints using the 3.0 logic that performs rank test before substitution. I think r028 is one of problematic requests.

The issue is that the errors are very imprecise for no good reason. Why can't we show which specific constraints are bad?

Evil-Spirit, 2017-04-06:
@whitequark 
I see allowed redundant constraints here, so I've found what 12.7 pt-pt-distances appied twice like as two pt-on-circle. If we are delete all the redundant, we can converge here fine.

Evil-Spirit, 2017-04-06:
@whitequark, Somehow too many redundant constraints can't be converged. I've figured out what 3xPT_ON_CIRCLE can't converge (just try to make it on empty sketch). After disallowing redundant, too many redundants threated as non-converged for no reason. This is probably bug, but I am not sure whether rank test can work fine or not for such amount of redundants.

Evil-Spirit, 2017-04-06:
@whitequark, I see allowed redundants here. The problem is probably the same as #225.

Evil-Spirit, 2017-04-06:
@whitequark, I've made a visualization for Newton steps. This is the hell https://youtu.be/72Dnqvn_uTk