solvespace/solvespace #227: Show units of dimensions in more precise way

Evil-Spirit, 2017-04-06:
This commit introduces units in meters https://github.com/Evil-Spirit/solvespace-master/commit/bc3e7a2e11d5e363d24d224b31df00013a996dfe
This https://github.com/Evil-Spirit/solvespace-master/commit/202c6fbb94297b0e632076dd22a5eeed47b6fba1 is experiments with si prefixes and formatting

whitequark, 2018-07-12:
Thank you for this excellent feature! I've merged the "dimensions in meters" patch in master. Do you think you could add "Dimensions with _S_I prefixes" as another menu option? This seems to be more in line with the rest of the options.

Evil-Spirit, 2018-07-12:
You mean to make this feature as option to have ability to switch it off?

whitequark, 2018-07-12:
I mean it makes sense to have separate options for "dimensions in mm" (for ISO technical drawings), "dimensions in m" (for architectural drawings) and "dimensions in metric with SI prefixes" (for unusual drawings where very large, very small or very different units are present). This is because most technical drawings have all length dimensions with the same units to make it less likely that e.g. someone misreads "mm" as "m" or maybe "cm".

Symbian9, 2018-07-28:
@whitequark, would you also merge "don't show zeroes" option?

whitequark, 2019-05-24:
@Evil-Spirit Thank you, merged in bc0e9c5, with many changes:
  * Added cm;
  * Added imperial units (mil, µin);
  * Changed formatting to use round-to-even, to match labels displayed with SI units off;
  * Adjusted measurement tools to always use SI prefixes, because 400000000 mm³ is not a useful value;
  * Overall simplification.

closed 2019-05-24 by whitequark