solvespace/solvespace #229: Circle radius can be negative

Evil-Spirit, 2017-04-06:
[image]

whitequark, 2017-04-06:
I don't think this is a bug. We allow all kinds of degenerate constraints, like line segments with length 0, and so on. It seems odd to prohibit those in some cases (like when specified explicitly) but not others (when they're inferred from other constraints).

Evil-Spirit, 2017-04-06:
@whitequark but this must show circle curve at least

whitequark, 2017-04-06:
Wait. How did you even manage to make a circle with negative radius?

Evil-Spirit, 2017-04-06:
I made two pt-on-circle and drag them in some unusal way

whitequark, 2017-04-06:
Oh, this is basically the same problem as constraint chirality, except applied to a circle.

whitequark, 2017-04-06:
We should probably rather fix this on the solver level, so that this is not possible.

Evil-Spirit, 2017-04-06:
@whitequark yes, the equation for pt-on-circle can be rewritten
`AddEq(l, du->Square()->Plus(dv->Square())->Sqrt()->Minus(r), 0);`

Evil-Spirit, 2017-04-06:
instead of 
`AddEq(l, ((du->Square())->Plus(dv->Square()))->Minus(r->Square()), 0);`

whitequark, 2017-04-08:
Fixed in master.

closed 2017-04-08 by whitequark