solvespace/solvespace #231: Build Fails to link on VS2015 Win10 x64

Jonimoose, 2017-04-07:
### System information

SolveSpace version: Git master (8fd11f48869a7d9d884da91055f14fd37d053fa1)

Operating system: Windows 10 64bit, fully updated as of 4/6/2017

### Expected behavior

What should have happened?
The project to compile in VS2015 using the instructions listed in the readme.

### Actual behavior

What actually happened?
Received the following errors:
1>------ Build started: Project: solvespace, Configuration: Debug x64 ------
1>zlibstaticd.lib(inflate.obj) : error LNK2019: unresolved external symbol inflate_fast referenced in function inflate
1>zlibstaticd.lib(deflate.obj) : error LNK2019: unresolved external symbol match_init referenced in function lm_init
1>zlibstaticd.lib(deflate.obj) : error LNK2019: unresolved external symbol longest_match referenced in function deflate_fast
1>F:\caring\Projects\solvespace\build\bin\Debug\solvespace.exe : fatal error LNK1120: 3 unresolved externals
========== Build: 0 succeeded, 1 failed, 15 up-to-date, 0 skipped ==========

whitequark, 2017-04-07:
Oh, this is because you're performing an x64 build. This configuration is neither tested nor currently supported (mostly because I never had a reason to).

whitequark, 2017-04-07:
To be clear, I'll fix this issue now, but for now there are no plans to build x64 builds on CI because they're less portable than x32 builds and, with other limits kicking in much earlier than address space size in SolveSpace, offer no real advantage.

whitequark, 2017-04-07:
Unable to reproduce. Just set up a fresh build directory using Visual Studio 14 2015 on Windows 10 x64, and the build succeeded. I'm not sure what I can do in regards to this report, so closing.

I can suggest trying to:
  * build a fresh tree (`git clean -dxf` `del /QS build`)
  * build with msbuild and not nmake (`cmake -G "Visual Studio 14 2015 Win64" ..` `cmake --build .`)

Jonimoose, 2017-04-07:
whitequark, The nmake 64bit build worked fine but didn't run do to another already reported issue. I received the errors in the first comment https://github.com/solvespace/solvespace/issues/231#issue-220070712 when building the project in visual Studio. I will attempt the listed commands this weekend and see if the issue disappears on my end.

whitequark, 2017-04-07:
@Jonimoose okay, I was confused by this part:

> What should have happened?
> The project to compile in VS2015 using the instructions listed in the readme.

There are a few different ways to build the project with VS tools, so I specifically tested nmake. Let me try outright VS now.

whitequark, 2017-04-07:
Nope, still unable to reproduce. I did:
```
cmake .. -G "Visual Studio 14 2015 Win64"
solvespace.sln
```
then just hit Ctrl+B.

closed 2017-04-07 by whitequark

Jonimoose, 2017-04-11:
Something must have been indeed wrong on my end. Build works correctly now. Thank you for your assistance on this, sorry it ended up being an issue on my end.

ssabpisa, 2017-12-02:
Using the same exact configuration:
- Checked out 8fd11f4
- Windows 10 64 bit
- Visual Studio 14 2015 Win64 (also tried Visual Studio 14 2015)
- Run ALL_BUILD in Release (also tried Debug)

 I was able to build successfully but none of the configuration ran.

whitequark, 2017-12-02:
@ssabpisa Apply the patch from #92.

ssabpisa, 2017-12-03:
Wow that worked. Thanks.