solvespace/solvespace #232: Problems with builing on Debian 8

Evil-Spirit, 2017-04-07:
Attempt to build on Debian 8 has filed,  gtkmm-3.16 is missing
…
— Checking for modules 'gtkmm-3.0>=3.16;pangomm-1.4;x11'
— 
CMake Error at /usr/local/share/cmake-3.7/Modules/FindPkgConfig.cmake:415 (message):
A required package was not found
Call Stack (most recent call first):
/usr/local/share/cmake-3.7/Modules/FindPkgConfig.cmake:588 (_pkg_check_modules_internal)
CMakeLists.txt:194 (pkg_check_modules)

— Configuring incomplete, errors occurred!
…

In repository there is  gtkmm-2.4 but 3.0 is missing.

- ukhanov -

whitequark, 2017-04-07:
This is expected, and was extensively discussed before. SolveSpace cannot be built with Debian Jessie (stable, soon to be oldstable) but can be built with Debian Stretch (testing, soon to be stable).

whitequark, 2017-04-07:
To elaborate on this, Stretch is already [frozen](https://release.debian.org/) and should become a proper release any day now; there's no reason to not migrate to Stretch already, especially on a desktop system.

closed 2017-04-07 by whitequark

ukhanov, 2017-04-08:
Debian 8 is tagget as 'wheezy' not 'jessie'. 
Anyway, gtkmm-3.16 does not exist on its official site: 
http://www.gtkmm.org/en/news.html
"April 6, 2011: gtkmm 3.0.0 and glibmm 2.28.0 released, as per the GNOME schedule. gtkmm 3.0 wraps the API in GTK+ 3.0 and installs in parallel with the gtkmm-2.4 API."

whitequark, 2017-04-08:
@ukhanov [this one](https://packages.debian.org/jessie/libgtkmm-3.0-dev) is versioned 3.14.0 (and is too old); [this one](https://packages.debian.org/stretch/libgtkmm-3.0-dev) is versioned 3.22.0 (and is new enough). Somewhere in between there might or might not have been a 3.16.0 package. I was looking at gtkmm changelogs when writing the version requirement.

ukhanov, 2017-04-08:
Thank you for clarifying the issue. BTW all versions are tagget as 'experimental' or 'unstable'. Even the old 2.4 one.

whitequark, 2017-04-08:
I have no idea what's up with that. gtkmm is certainly stable enough for real work, and it even has a stable ABI.