solvespace/solvespace #237: Creating a solid bounded by wireframe

whitequark, 2017-04-16:
Right now, there are certain classes of solids that are extremely tedious or completely impossible to define in SolveSpace. These include:
  * platonic solids (except cubes)
  * Archimedean solids
  * stellations
  * frustums
  * certain pyramids
and so on.

Some of these could clearly be done in the spirit of SolveSpace (for example, a frustum group would cover right pyramids and frustums). However, many others aren't. They are needed somewhat rarely, but when they *are* needed, you are essentially out of luck right now. If you could import STEP, which you will be able to in not-so-near future, then that would not be a show-stopper, but it would still break parameterism completely.

Question is: can we add an escape hatch for this? Allow defining a solid with a set of lines and points; perhaps even drop curves. How hard is it to integrate with our NURBS backend? How hard would it be to define a reasonable UI? Is the operation even well-defined, in general?

cc @Evil-Spirit @jwesthues

jwesthues, 2017-04-16:
I don't think a useful "shell from wireframe" group is too hard. Take the lines/edges in 3d, and find all the planes defined by some set of adjacent edges. In each plane, find all the closed loops, and those are your faces. Then test that the faces make a watertight, non-self-intersecting shell, choose their normal directions to be consistent (like by choosing one arbitrarily, and then flooding out from its edges), and then make sure the shell isn't inside-out (like by computing its volume, and flipping all the normals if it's negative). Curves get trickier, but I doubt the plane version of this would be more than a few hundred lines of code.

Evil-Spirit, 2017-04-16:
@whitequark, This is looks like common triangle mesh editing mode like it done e. g. in 3d studio max. We are defining the common 3d entities(segments, circles, arcs) and introduce something like "3d polygon" request which creatse face from set of looped edges and perserve points to lay in the same plane using solver equations. This is like [direct modeling](http://isicad.ru/ru/news.php?news=12886) in the context of single group of SolveSpace's history based approach. We can also introduce watertight and solid checks in the same way like it done for checking closed contours.

whitequark, 2017-04-16:
Hm, this is a very interesting idea. We can add a "direct modeling group", where you would use parametric entities, but unlike regular draw-in-3d groups, the group would also include a solid, defined in the way @jwesthues described above. I think this solves all UI concerns I had.

phkahler, 2019-06-05:
Would it be OK to have a checkbox on sketch-in-3d groups that causes a shell to be created from the entities in the group? There would be limitations on what you can create a shell from, but the UI would just be a checkbox. This shouldn't be hard for straight edges and convex objects.

whitequark, 2019-06-11:
The restriction to convex objects seems quite unfortunate. Does it buy us much?

henrebotha, 2023-07-06:
I still hope to see this feature in SolveSpace. For most of my use cases, it would remove the need for lofts (e.g. lofting from one square to another is just a matter of drawing connecting lines and converting the whole thing to a solid). Even if it's restricted to planar faces only, it would still be extremely useful.

percontation, 2024-07-25:
Even the restricted form of turning 3d sketches into convex hulls would be incredibly useful. Many non-convex cases could be handled by this same feature just by subtracting a second convex hull.

I've wanted this feature for filling in gaps in models ("bridging" two edge loops together with a simple 3d transition shape) or, as henrebotha mentioned, an alternative to lofts in some simple cases.