solvespace/solvespace #243: Font oops,

romeojulietthotel, 2017-04-24:
### System information

SolveSpace version: git master

Operating system: Linux

### Expected behavior

What should have happened?
No oops.

What actually happened?

Was trying out the text feature and it oopsed.

### Additional information

freetype: cannot set character size: invalid pixel size
oops at line 261, file /b/b/solvespace/src/ttf.cpp

whitequark, 2017-04-25:
That's not git master. The line 61 of ttf.cpp in master does not have a call to `oops`. What is the exact git revision you're using?

romeojulietthotel, 2017-04-25:
I oopsed, I had switched to v2.3. I updated the issue.

romeojulietthotel, 2017-04-25:
git master 3.0~50789079

```
line 349, function AspectRatio:
Assertion 'Expected font face to be loaded' failed: 
((fontFace != NULL) == false)
```

I was trying the different fonts and when selecting a font I got that.

whitequark, 2017-04-25:
Which font? Also, post all messages, not just the crash.

romeojulietthotel, 2017-04-26:
There is more than one font that causes the assert. Here's one:
clR6x12.pcf a.k.a. Clean

whitequark, 2017-04-26:
Well, how do I obtain this font?

romeojulietthotel, 2017-04-26:
Depends on your os but on linux it's in a font package.

whitequark, 2017-04-26:
If you expect me to fix this, please provide the font.

romeojulietthotel, 2017-04-26:
Ok, for what linux distro?

whitequark, 2017-04-26:
I run Debian. You could also just provide the file.

romeojulietthotel, 2017-04-26:
https://packages.debian.org/jessie/all/xfonts-base/filelist

romeojulietthotel, 2017-04-26:
>If you expect me to fix this,

FWIW I don't expect anything. I try to offer help. Maybe you are suffering from those people that seem to think they are entitled. That's a trap I see maintainers fall into and it's not fair to the maintainer.

It takes time to file reports and it takes time to fix them. It also takes time to learn an application. I am doing what I can. Thanks for your efforts. Together we make the app better.

whitequark, 2017-04-26:
Okay, I have that package installed on my system, and I have `/usr/share/fonts/X11/misc/clR6x12.pcf.gz`. I cannot select it from the font list though, how did you achieve that? It seems that on my machine fontconfig does not return pcf fonts for some reason.

whitequark, 2017-04-26:
I found some other fonts that exhibit the same issue (they all start with "STIX" on my machine; not sure what are these fonts), and it is fixed as of 280d5b90.

closed 2017-04-26 by whitequark

romeojulietthotel, 2017-04-26:
FWIW, I think fc-cache has to be run for new fonts. But I'd expect that to happen on install.

romeojulietthotel, 2017-04-26:
This does not fix the issue.

whitequark, 2017-04-26:
Are you sure you've rebuilt the binaries? What is the complete content of messages printed to stdout?

romeojulietthotel, 2017-04-26:
Error now:
 ```
line 350, function AspectRatio:
Assertion 'Expected font face to be loaded' failed: ((fontFace != NULL) == false)

```

whitequark, 2017-04-27:
What is the complete content of messages printed to stdout? You've only posted the error, not the preceding messages...

whitequark, 2017-04-27:
> FWIW, I think fc-cache has to be run for new fonts.

I've tried rerunning fc-cache and it does nothing to make that font available for use.