solvespace/solvespace #244: Feature Request: Rotate camera around Up axis

tilya2014, 2017-04-25:
In most of 3d editors user can fix camera to rotate around one axis. Can somebody add this option to solvespace. I've changed this in my sources and its working, but I think that should be optional. 

Version: 3.0~7544d016
solvespace-core/mouse.cpp line 133
```
        ....
        Vector up = Vector::From(0,0,1);
        if(!(shiftDown || ctrlDown)) {
            double s = 0.3*(PI/180)*scale; // degrees per pixel
	    projRight = orig.projRight.RotatedAbout(up, -s*dx);
            projUp = orig.projUp.RotatedAbout(orig.projRight, s*dy);
           ....

```

whitequark, 2017-04-26:
Is this the same as #123?

tilya2014, 2017-04-27:
Yes it is. I didn't notice. Thanks

closed 2017-04-27 by whitequark