solvespace/solvespace #245: Add SVG import support

tpimh, 2017-04-27:
It is a feature request.

I find it nearly impossible to create a gear in solvespace. So I try generate a gear with an online tool (geargenerator.com or paulkaplan.me/Gears/), download SVG, edit it in Inkscape, then in order to get the outline of the gear into solvespace, I should convert it to DXF. It would be great to eliminate the last step and import SVG outline to the workplane.

whitequark, 2017-04-27:
FWIW, I find it more important to make gear creation easier in SolveSpace than to add SVG import support.

tpimh, 2017-04-27:
It would be great. I also tried creating screw thread and failed. Is it even possible now?

whitequark, 2017-04-27:
Gears, yes, sure: http://solvespace.com/forum?action=viewthread&parent=872

Screws, no, but our NURBS backend simply does not support the required operation. You could import an external STEP model if we added that (#88).

hinell, 2022-01-11:
Hi. Web developer is here.

I can tell that it shouldn't be hard to parse and import SVG data. Though I agree that it's simpler to implement a procedural gear generator.

The most sophisticated element in SVG is the [`<path d="...." />`][path]. It has [d attribute] that basically directs line drawing by simple commands like move (M, m), lineTo (L, l) etc.

[d attribute]: https://developer.mozilla.org/en-US/docs/Web/SVG/Attribute/d
[path]: https://developer.mozilla.org/en-US/docs/Web/SVG/Element/path

hinell, 2022-01-12:
I've found that it's possible to export `DFX` from `SVG` by using [Inkscape](https://inkscape.org/). The latter has compatible GPL3 license btw.

ruevs, 2022-01-12:
The gear link:
https://solvespace.com/forum.pl?action=viewthread&parent=872

ruevs, 2022-01-12:
> I've found that it's possible to export DFX from SVG by using Inkscape. The latter has compatible GPL3 license btw.

DXF, absolutely! And it works. For simple SVG files the exported DXF if "properly" imported in SolveSpace.

And:
- SVG is not a CAD file format - so any import we implement will be "fuzzy and incomplete" and someone will complain.
- SVG is a huge and complicated format.
Even if we use an existing library we will drag in a significant amount of... weight... if we support it.
- SolveSpace is not a file format converter. This job is better handled by [Inkscape](https://inkscape.org/) - the "native" editor for SVG.

So I will close this.

closed 2022-01-12 by ruevs

tpimh, 2022-01-13:
Understandable decision with very good argumentation. I'm converting all my SVGs to DXFs now.