solvespace/solvespace #247: background color of property window

tgirod, 2017-04-29:
Property window's background is black, same as main window's background. Without a border around the property window, it is hard to distinguish the two windows from each others.

[image]

As a result, I often find myself clicking on the property window when trying to move the view around, which is annoying.

Adding a small border around the property window would help, or maybe changing the background color ...

whitequark, 2017-04-29:
> Adding a small border around the property window would help

On Linux, this is the responsibility of the window manager. In my case, the window manager, i3, draws a nice thick border:
[image]

Therefore I suggest you configure your window manager to do the same. I don't know what DE that is or how does one configure it that way.

tgirod, 2017-04-29:
fair enough !

closed 2017-04-29 by whitequark