solvespace/solvespace #249: Sometimes solvespace creates corrupt slvs files.

nekromant, 2017-04-30:
### System information

This is SolveSpace version 2.3~.
(Latest deb-packaged version from releases)

Operating system: Debian Jessie (stable)
 
### Expected behavior

Solvespace should always save files correctly and load them. 

### Actual behavior

Sometimes, when saving complex designs solvespace produces a file it can't open afterwards. Opening such a corrupt file causes solvespace's window to stop responding and the solvespace process starts consuming 100% of CPU.  Removing some of the groups by hand from the slvs file sometimes helps out, but it's hard to guess which group is causing problems if there are many.  

### Additional information

For some reason github doesn't allow me to upload a zipped slvs, so here's a link to the corrupt slvs on my owncloud. 

https://cloud.ncrmnt.org/index.php/s/5qM4MDnD7suMwxs

whitequark, 2017-04-30:
The file is definitely not corrupted. Both 2.3 and 3.0 can open the file, after a slight delay:
```
Generate::REGEN (for bounding box) took 50 ms
Generate::REGEN took 99 ms
Generate::ALL (for bounding box) took 416 ms
didn't converge
have 49.789 62.768 17.005
want 48.464 62.768 19.087
distance = 2.46811
Generate::ALL took 6478 ms
Generate::DIRTY (for bounding box) took 52 ms
Generate::DIRTY took 2387 ms
```
Is, perhaps, chord tolerance set to an unreasonably low value on your machine?

nekromant, 2017-04-30:
@whitequark Hm, I think I figured it out. My Intel Sandybridge drivers were the issue. Since I'm using a dual-gfx laptop, running solvespace using the nvidia card solved the issue for me. 

Here's what I get from stdout from solvespace when trying to open the file on intel :
```
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 14: reading configurations from ~/.fonts.conf is deprecated. please move it to /home/necromant/.config/fontconfig/fonts.conf manually
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
The program 'solvespace' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadMatch (invalid parameter attributes)'.
  (Details: serial 5291 error_code 8 request_code 72 minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

And here's what I get from gdb when I ctrl+c it from the frozen state: 

```
Program received signal SIGINT, Interrupt.
0x00007ffff4b842cd in _XReply () from /usr/lib/x86_64-linux-gnu/libX11.so.6
(gdb) bt
#0  0x00007ffff4b842cd in _XReply () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#1  0x00007ffff4b68d6e in XGetGeometry () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#2  0x00007ffff76fd045 in ?? () from /usr/lib/x86_64-linux-gnu/libGL.so.1
#3  0x00007fffe36ec8f7 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#4  0x00007fffe36eb712 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#5  0x00007fffe3631b2e in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#6  0x00007fffe3632d48 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#7  0x00007fffe36024d0 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#8  0x00007fffe3616af5 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#9  0x00007fffe35eb076 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#10 0x00007fffe35d222c in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#11 0x00007fffe35e80b4 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#12 0x00007fffe34da933 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#13 0x00007fffe34daa18 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#14 0x00007fffe36328bb in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#15 0x00007fffe36eb41f in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#16 0x00007fffe36e8755 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#17 0x00007ffff76d5545 in glXMakeCurrentReadSGI () from /usr/lib/x86_64-linux-gnu/libGL.so.1
#18 0x0000000000463325 in SolveSpace::TextWindowGtk::~TextWindowGtk() ()
#19 0x00007ffff3b14b29 in __run_exit_handlers (status=1, listp=0x7ffff3e825a8 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true) at exit.c:82
#20 0x00007ffff3b14b75 in __GI_exit (status=<optimized out>) at exit.c:104
#21 0x00007ffff5c43001 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#22 0x00007ffff4b8645d in _XError () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#23 0x00007ffff4b833c7 in ?? () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#24 0x00007ffff4b83485 in ?? () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#25 0x00007ffff4b84430 in _XReply () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#26 0x00007ffff4b68d6e in XGetGeometry () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#27 0x00007ffff76fd045 in ?? () from /usr/lib/x86_64-linux-gnu/libGL.so.1
#28 0x00007fffe36ec8f7 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#29 0x00007fffe36eb712 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#30 0x00007fffe3631b2e in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#31 0x00007fffe3632d48 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#32 0x00007fffe36024d0 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#33 0x00007fffe3616af5 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
---Type <return> to continue, or q <return> to quit---
#34 0x00007fffe35eb076 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#35 0x00007fffe35cf7a7 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#36 0x00007fffe35cfb96 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#37 0x00007fffe35d013e in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#38 0x00007fffe35d1739 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
[... 130 lines cut ...]
    GL_EXT_shader_integer_mix, GL_EXT_shadow_funcs, GL_EXT_stencil_two_side, 
    GL_EXT_stencil_wrap, GL_EXT_subtexture, GL_EXT_texture, GL_EXT_texture3D, 
    GL_EXT_texture_array, GL_EXT_texture_compression_latc, 
    GL_EXT_texture_compression_rgtc, GL_EXT_texture_cube_map, 
    GL_EXT_texture_edge_clamp, GL_EXT_texture_env_add, 
    GL_EXT_texture_env_combine, GL_EXT_texture_env_dot3, 
    GL_EXT_texture_integer, GL_EXT_texture_lod_bias, 
    GL_EXT_texture_mirror_clamp, GL_EXT_texture_object, 
    GL_EXT_texture_rectangle, GL_EXT_texture_sRGB, GL_EXT_texture_sRGB_decode, 
[... cut ...]
    GL_NV_depth_clamp, GL_NV_fog_distance, GL_NV_light_max_exponent, 
    GL_NV_packed_depth_stencil, GL_NV_primitive_restart, 
    GL_NV_texgen_reflection, GL_NV_texture_env_combine4, 
    GL_NV_texture_rectangle, GL_OES_EGL_image, GL_OES_read_format, 
    GL_SGIS_generate_mipmap, GL_SGIS_texture_border_clamp, 
    GL_SGIS_texture_edge_clamp, GL_SGIS_texture_lod, GL_SUN_multi_draw_arrays
``` 

Still not sure whether it's an intel drivers bug or a solvespace bug, so reporting everything here. Hope it helps.

whitequark, 2017-04-30:
Neither. You aren't actually running SolveSpace on the Intel video card. You are using the Mesa/LLVMpipe software rasterizer (and I'd expect that to be noticeably slow, if not *unusably* slow).

See:

    OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.5, 256 bits)

And from the backtrace:

    #3  0x00007fffe36ec8f7 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so

The Mesa software rasterizer is known to misbehave when used with SolveSpace, although I never figured out why. It is not considered a supported configuration.

nekromant, 2017-04-30:
Okay, looks like we can close the issue for now. I'll have to tripple check my graphics settings, looks like I've got something terribly broken without me noticing.

closed 2017-04-30 by nekromant