solvespace/solvespace #250: Make two rows for the icons in the Property Browser.

romeojulietthotel, 2017-04-30:
Enhancement request:

Make two rows for the icons in the Property Browser so the window can be narrow.

I looked at textwin.cpp but not sure how to best implement this feature.

On a low resolution display (laptop) the horizontal display space is limited and having a narrower Property Browser would give more to the modeling area.

whitequark, 2017-04-30:
We really should wrap both text and buttons for such cases.

romeojulietthotel, 2017-05-04:
Having a resizable window would be good to have.  I can only guess, since I don't know, that GTK has some features to do this so no hard coded geometry/locations would be needed.

And avoiding hard coded values in code is more maintainable as time goes by.

whitequark, 2017-05-04:
GTK has nothing to do with this because all of our UI drawing code is in OpenGL.

> And avoiding hard coded values in code is more maintainable as time goes by.

I'm well aware. I did not write that code.

romeojulietthotel, 2017-05-04:
Not too surprised that I was wrong about the GUI. It's good that it doesn't use a big UI toolkit.
Now I know why it compiles so quickly and with very few dependencies.