solvespace/solvespace #251: Solvespace crash if I try to lathe a shape in "difference" mode

patrovite, 2017-05-03:
### System information

SolveSpace version: Solvespace 3.0 - 6637

Operating system: Voyager linux =>  XUbuntu 16.04 LTS

### Expected behavior

On a solid I tried to do a hole with a chamfer

### Actual behavior

On the workplane I draw the shape and lathe it. The program crashed when I selected "Difference" in the lathe options

### Additional information

[test3.zip](https://github.com/solvespace/solvespace/files/974740/test3.zip)

patrovite, 2017-05-03:
If I launch Solvespave from the console, when the program crashed it produced these messages : 

File /home/pierre/src/solvespace/src/dsc.h, line 334, function FindById: 
Assertion 'Cannot find handle' failed: ((t != NULL) == false). 
Backtrace: 
0: solvespace(_ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_+0x58) [0x483be8] 
1: solvespace(_ZN10SolveSpace6Vector4FromENS_6hParamES1_S1_+0x77) [0x5671d7] 
2: solvespace(_ZN10SolveSpace5Group18ExtrusionGetVectorEv+0x39) [0x4e6489] 
3: solvespace(_ZN10SolveSpace10TextWindow23ScreenChangeGroupOptionEij+0x132) [0x551b22] 
...

whitequark, 2017-05-04:
This was fixed in 85b4b25278c0acf85c6f35cb347c672f273dec5d, are you sure your version is up-to-date?

patrovite, 2017-05-04:
I downloaded and compile my version in april 2th. I'm probably not up to date. I'll will dowload and compile the latest github version.

patrovite, 2017-05-04:
With the last version it solved my problem.
Thanks

closed 2017-05-04 by patrovite