solvespace/solvespace #253: weird behaviour with imported dxf

tgirod, 2017-05-04:
### System information

SolveSpace version:  2.3~7c1ca460

Operating system: archlinux

### All is fine ...

I'm importing a vectorized (dxf) tree. The shape has nice closed contours and everything is fine.

[image]

### But then ...

Now, if I add a rectangle around this shape, I should get the negative shape, right ? But this is what I get instead.

[image]

It makes no sense to me. Here is the culprit.

[arbre.slvs](https://github.com/solvespace/solvespace/files/977430/arbre.txt)

whitequark, 2017-05-04:
That's because the style of the inner lines is different from the style of the rectangle (as we import DXF styles as they are). I'm not sure what's best to do here...

tgirod, 2017-05-05:
OK, I've just discovered the existence of styles in SolveSpace 👍 

Knowing this, it was trivial to solve the problem by selecting everything and removing assigned style. Thanks !

closed 2017-05-05 by tgirod