solvespace/solvespace #255: Program doesn't start after adding a new translation

patrovite, 2017-05-07:
### System information

SolveSpace version: 3.0~280d5b90

Operating system: xubuntu 16.04LTS

### Expected behavior

The program must start and the main windows displayed. 

### Actual behavior

The program doesn't start.
In the console I have :
File /home/pierre/src/solvespace/src/resource.cpp, line 496, function ExpectRegex:
Assertion 'Unmatched regex' failed: ((TryRegex(re, m)) == false).
Backtrace:
 0: ./solvespace(_ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_+0x58) [0x483f28]
 1: ./solvespace(_ZN10SolveSpace7LocalesEv+0x7ef) [0x53121f]
 2: ./solvespace() [0x53135c]
 3: ./solvespace(_ZN10SolveSpace9SetLocaleERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x9) [0x531ac9]
 4: ./solvespace(main+0x30d) [0x4587fd]
 5: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f666f0b3830]
 6: ./solvespace(_start+0x29) [0x45ad09]
Abandon

### Additional information

This problem appear after I added a new translation.
I added "fr-FR.po" in "res/locales" and in "res/locales.txt" I added the line "fr-FR,040C,Français" (I tried with "040c"). After I did a "make" in the build dir.
[french translation issue.zip](https://github.com/solvespace/solvespace/files/981917/french.translation.issue.zip)

whitequark, 2017-05-07:
My bad. I've fixed it in master.

closed 2017-05-07 by whitequark

Symbian9, 2017-07-12:
@patrovite, try make DEB from latest sources of SolveSpace 3.0 for xubuntu 16.04 and upload it somewhere, please!