solvespace/solvespace #256: Cannot make circle from two arcs and three pts-coincident constraints

shrikumarh, 2017-05-08:
### System information
solvespace 2.3+repack1-2~bpo8+1
Operating system: (e.g. Debian testing) debian/jessie

### Expected behavior

I should be able to make a circle by melding together two arcs.

### Actual behavior

If I use constraints to meld two arcs together, it fails, in different ways. 

I can do it by mousing one center to the other just fine. 

Sometimes I can do it by setting separation distance == 0. But somtimes not. 

But I cant evet get the solver to join the centers together by pts-coincident.

What actually happened?

Please see  attached solvespace file, and reproduce this:

1. Move center 1 to center 2 by mouse. It appears to work as expected except the dots are not truly coincident ... the other dot keeps running away instead of ever turning yellow.
2. Move center 1 to center 2 by making points-coincident .. it bombs as Unsolvable.
3. Move center 1 to center 2 by setting distance between points to 0mm .. Sometimes it works, sometimes it bombs as Redundant Constraints. (also tried with Step Dimension)
4. These above are simpler minimalized cases of a problem I stumbled upon while trying to draw a Reuleaux Triangle (a triangular wheel) When you try to constrain the arc centers to any of the triangle vertices, it bombs out in different ways under different sequences of operations.

I had also tried these separately on an empty sketch, with no superflous adds, deletes and edits .. just draw two arcs, constrain end 1 to end 1. Then end 2 to end 2. Then constrain their centers coincident.
Boom!

### Additional information

untitled.slvs attached.
[reuleaux-triangle.zip](https://github.com/solvespace/solvespace/files/982341/reuleaux-triangle.zip)

whitequark, 2017-05-08:
The problem here is that you're overconstraining the sketch. Once you get the "SOLVE FAILED! redundant constraints" error message, click the "allow redundant constraints" link in the property browser.

> Move center 1 to center 2 by setting distance between points to 0mm .. Sometimes it works, sometimes it bombs as Redundant Constraints. (also tried with Step Dimension)

The fact that it *works* here is a bug; it should always error out as the sketch is overconstrained. This is fixed in master.

shrikumarh, 2017-05-14:
Thanks much. I understand what you mean, but to reason out what inside made it overconstrained was a bit confusing. I scratched my head a good bit about this one, but think I finally stumbled upon a few brain cells somewhere inside my skull.

Maybe the following is what you are saying is what's happening?
1. Two arcs, with the two ends constrained touching end1 to end1, end2 to end2. The two centers have one degree of freedom .. they can slide on the center line between the two arc-end-points.
2. When you mark the two centers with a distance measure constraint and set that distance measure = 0, you are removing exactly that one degree of freedom. Thats works.
3. When you join the two centers with a pts-coincident constraint, then internally it is setting x1=x2, and y1=y2, which is trying to remove two degrees of freedom. Either one of x1=x2 or y1=y2 would suffice. Hence redundant constraints.

Wonder if this is a correct explanation .. Is it?
This was at first not obvious to me and my kids when we were coming from the intuition which is embodied in the UI. (without diving under the covers into the math).

If this is correct, then some improvement in the UI showing degrees of freedom may perhaps help? 

Thinking aloud some suggestions: Currently, when you click  "Show Degrees of Freedom", it highlights points that can move. But if there are three points highlighted, that's not mean three "Degrees of Freedom"! That only means there are three points that are "Not Fully Constrained"

Maybe another way to Show Degrees of Freedom might be to show (short segments of) lines on orthogonal axes at each point which can move in certain (arbitrary) axes. A point with one degree of freedom will have one arrow, and one with two degrees of freedom will show two orthogonal arrows. (of course, there are many different orthogonal pairs of arrows possible ... pick one arbitrarily).

Also, lets say there are two points that can move, but each one is dependent on the other .. if you move one, you are fully constraining they other. (eg. line ends with midpoint fixed on sheet origin). Then only one point should be highlighted as a Show Degrees of Freedom, showing one arrow, along the line. Maybe other point can be highlighted when asked to Show an Alternative Degree of Freedom. (and so on . cycle if there are more than two points that are not fully constrained).

Currently whats called Show Degrees of Freedom is really not degrees of freedom, but really points that are (taken individually in isolation) "Points Not Fully Constrained". Three points that are not fully constrained does not mean there are three degrees of freedom. But the UI menu label "Show Degrees of Freedom" suggests that incorrect meaning very strongly.

Just a suggestion, hope it is useful.

-- //Shrikumar

whitequark, 2019-05-24:
@jwesthues What do you think about renaming "Show Degrees of Freedom" to "Show Underconstrained Points" or something like it?

jwesthues, 2019-05-24:
Sounds reasonable to me, indeed perhaps clearer.

whitequark, 2019-05-24:
Fixed in 88879d3.

closed 2019-05-24 by whitequark