solvespace/solvespace #258: Remove hard coded text in order allow translation

patrovite, 2017-05-08:
### System information

SolveSpace version: 3.0....
Operating system: XUbuntu 16.04LTS

### Expected behavior

With French translation all the text must be French.

### Actual behavior

After the French translation has been finished texts in the property browser are still in english

### Additional information

whitequark, 2017-05-08:
Yes, the much needed property browser rendering rework...

patrovite, 2017-05-20:
Are you planning to do a global rework on the property browser or you only want to work on the hard coded text? If it's the second choice, I can help you.

whitequark, 2017-05-20:
The former. There are more bugs like https://github.com/solvespace/solvespace/issues/92 and https://github.com/solvespace/solvespace/issues/250 and others maybe...

closed 2022-07-05 by ruevs