solvespace/solvespace #259: 'Cannot allocate memory' when exporting mesh

jonas-schievink, 2017-05-20:
### System information

SolveSpace version: 3.0~33b6e517 (current master, built with clang 4.0)

Operating system: Arch Linux x86-64

### Expected behavior

Solvespace should have reported an error (since I'm pretty certain the model has issues that lead to a somewhat broken mesh) or it should successfully export the mesh.

### Actual behavior

Solvespace hangs for a while, then crashes with a failed 'Cannot allocate memory' assertion.

### Additional information

```
File ../../src/platform/unixutil.cpp, line 105, function MemAlloc:
Assertion 'Cannot allocate memory' failed: ((p != NULL) == false).
Backtrace:
 0: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_+0x50) [0x48ad60]
 1: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace() [0x48aeec]
 2: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace7SKdNode12SnapToVertexENS_6VectorEPNS_5SMeshE+0x3f0) [0x4f6e30]
 3: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace7SKdNode12SnapToVertexENS_6VectorEPNS_5SMeshE+0xc2) [0x4f6b02]
 4: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace7SKdNode12SnapToVertexENS_6VectorEPNS_5SMeshE+0x82) [0x4f6ac2]
 5: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace7SKdNode12SnapToVertexENS_6VectorEPNS_5SMeshE+0xc2) [0x4f6b02]
 6: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace7SKdNode12SnapToVertexENS_6VectorEPNS_5SMeshE+0xc2) [0x4f6b02]
 7: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace7SKdNode12SnapToVertexENS_6VectorEPNS_5SMeshE+0x82) [0x4f6ac2]
 8: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace7SKdNode12SnapToVertexENS_6VectorEPNS_5SMeshE+0xc2) [0x4f6b02]
 9: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace7SKdNode12SnapToVertexENS_6VectorEPNS_5SMeshE+0xc2) [0x4f6b02]
10: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace7SKdNode10SnapToMeshEPNS_5SMeshE+0x1ec) [0x4f397c]
11: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace5Group20GenerateShellAndMeshEv+0x1475) [0x4e2005]
12: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace12SolveSpaceUI11GenerateAllENS0_8GenerateEbb+0x8eb) [0x4d1ccb]
13: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace12SolveSpaceUI12ExportMeshToERKNS_8Platform4PathE+0x2b) [0x4b2deb]
14: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_ZN10SolveSpace12SolveSpaceUI8MenuFileENS_7CommandE+0x88b) [0x46d77b]
15: /usr/lib/libgtkmm-3.0.so.1(_ZN3Gtk14MenuItem_Class17activate_callbackEP12_GtkMenuItem+0x50) [0x7f95839c8810]
16: /usr/lib/libgobject-2.0.so.0(g_closure_invoke+0x145) [0x7f9581cbef75]
17: /usr/lib/libgobject-2.0.so.0(+0x2195c) [0x7f9581cd095c]
18: /usr/lib/libgobject-2.0.so.0(g_signal_emit_valist+0xe3c) [0x7f9581cd9bdc]
19: /usr/lib/libgobject-2.0.so.0(g_signal_emit+0x8f) [0x7f9581cd9fbf]
20: /usr/lib/libgtk-3.so.0(gtk_widget_activate+0x6e) [0x7f9582c2dafe]
21: /usr/lib/libgtk-3.so.0(gtk_menu_shell_activate_item+0x106) [0x7f9582afb9f6]
22: /usr/lib/libgtk-3.so.0(+0x24cd2b) [0x7f9582afbd2b]
23: /usr/lib/libgtk-3.so.0(+0x22fcd1) [0x7f9582adecd1]
24: /usr/lib/libgobject-2.0.so.0(+0x101a4) [0x7f9581cbf1a4]
25: /usr/lib/libgobject-2.0.so.0(g_signal_emit_valist+0x601) [0x7f9581cd93a1]
26: /usr/lib/libgobject-2.0.so.0(g_signal_emit+0x8f) [0x7f9581cd9fbf]
27: /usr/lib/libgtk-3.so.0(+0x37c5fc) [0x7f9582c2b5fc]
28: /usr/lib/libgtk-3.so.0(+0x22ccae) [0x7f9582adbcae]
29: /usr/lib/libgtk-3.so.0(gtk_main_do_event+0x79e) [0x7f9582addd1e]
30: /usr/lib/libgdk-3.so.0(+0x36d65) [0x7f95825efd65]
31: /usr/lib/libgdk-3.so.0(+0x67df2) [0x7f9582620df2]
32: /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f95819e57b7]
33: /usr/lib/libglib-2.0.so.0(+0x4aa20) [0x7f95819e5a20]
34: /usr/lib/libglib-2.0.so.0(g_main_loop_run+0xc2) [0x7f95819e5d42]
35: /usr/lib/libgtk-3.so.0(gtk_main+0x85) [0x7f9582adced5]
36: /usr/lib/libgtkmm-3.0.so.1(_ZN3Gtk4Main3runERNS_6WindowE+0x121) [0x7f95839c4fb1]
37: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(main+0x661) [0x475fe1]
38: /usr/lib/libc.so.6(__libc_start_main+0xf1) [0x7f9580202511]
39: /home/jonas/dev/solvespace/build/solvespace-Desktop_Clang-Release/bin/solvespace(_start+0x2a) [0x46941a]
The program has unexpectedly finished.
```

Open [this file](https://github.com/solvespace/solvespace/files/1016276/case.zip), click `File -> Export Triangle Mesh...` (doesn't matter which file type you choose).

whitequark, 2017-05-20:
What's the export chord tolerance? You could genuinely have a mesh larger than ~3GB.

jonas-schievink, 2017-05-20:
0.01mm. Looks like 0.1mm works. However, a previous version of this model exported just fine at 0.01mm and resulted in an STL file of only about 140 KiB. The only difference between the versions are slightly changed constraints.

whitequark, 2017-05-20:
Render chord tolerance is in % not mm.

jonas-schievink, 2017-05-20:
But the calculated value in mm is displayed next to the value in %, no? Setting the percentage such that this value is 0.01mm still works (which incidentally is 0.01%). Even with 0.005% it still renders fine (and displays a rounded tolerance of 0.00mm).

I also get the feeling that the GUI is a bit borked since the displayed values are sometimes not updated and sometimes change when I change nothing, so it's a bit hard to tell what the *actual* values are.

Evil-Spirit, 2017-05-22:
@whitequark, @jonas-schievink fixed here

whitequark, 2017-05-22:
@Evil-Spirit Can you check if this addresses any of our other issues where mesh generation is slow?

Evil-Spirit, 2017-05-22:
@whitequark ofc

whitequark, 2018-07-12:
@Evil-Spirit Thanks!

Fixed in master.

closed 2018-07-12 by whitequark