solvespace/solvespace #260: [Feature request] Ability to move entities with keyboard

Helios747, 2017-05-24:
So a lot of my workflows across my system for various reasons are very keyboard centric. What I would find nice to have is the ability to move selected entities via the keyboard arrows or through some shortcuts. If it's a single object, move that, if it's a group selection, move them all.

When I had the grid enabled, my intuition told me that I could use arrow keys and some sort of key modifier to move what I had selected and it would move by points on the grid.

whitequark, 2017-05-24:
@jwesthues So this is an accessibility issue. SolveSpace is already much more accessible than direct CADs (or so I'm told), so I'd quite like to make it even better. Any opinion on the modifier? Ctrl+arrows, then jump using the current grid step?

jwesthues, 2017-05-24:
Nudge with the arrow keys is pretty common, makes sense.

Do you need a modifier? The arrow keys alone don't do anything yet, and that's how Office etc. work.

And maybe nudge by a grid step if the grid is shown, otherwise by a constant distance in screen pixels? That would avoid surprising users who haven't set up a grid (which is probably a lot of them, since the workflow isn't generally too grid-centric).

Evil-Spirit, 2017-05-24:
@jwesthues 
I think by default arrows must control camera/view. This is needed when you are controlling without mouse (I don't like touch pad so much)

jwesthues, 2017-05-24:
Are there other applications where arrow keys alone pan/rotate the view, and modifier+arrow moves entities? The reverse would have seemed more intuitive to me, but I can't immediately think of examples of either.

whitequark, 2017-05-24:
> Are there other applications where arrow keys alone pan/rotate the view, and modifier+arrow moves entities?

Photoshop moves selection with plain arrows, Inkscape moves the canvas. I don't see either choice as more intuitive, personally, and in either case it seems trivial to adjust to it. We already have a highly nonstandard mouse controls for camera and it's not like many people complain.

@Helios747 any opinion here?

Helios747, 2017-05-24:
I think with no modifier, moving the camera is fine.

Symbian9, 2017-07-12:
> Are there other applications where arrow keys alone pan/rotate the view, and modifier+arrow moves entities? The reverse would have seemed more intuitive to me, but I can't immediately think of examples of either.

Maybe, better way is add "Keyboard settings" for SolveSpace, where user will map keys as want?

whitequark, 2018-09-27:
@marthinwurer A very substantial amount of work is required before that's feasible.

marthinwurer, 2018-09-27:
@whitequark Yeah, I realize that. I've been working my way through the codebase. It doesn't hurt to dream big, though :)

baryluk, 2020-01-16:
I do have an issue of moving some entities, especially line segment and linear dimension that are overlapping. I am able to select the dimension, but then if I try to drag it nothing happens. Having ability to 1) disambiguate selection, 2) move selected entity by keyboard, would be somehow nice in screen space x-y coordinates.

Evil-Spirit, 2020-01-24:
> I do have an issue of moving some entities, especially line segment and linear dimension that are overlapping.

SolveSpace should use the two approaches (my therminology):
1. Selection based on Voroni - just pick the entitiy with the minimum distance from mouse pointer.
2. Select-by-order - if nearest entity already selected, pick the next nearest one.

Probably, selection by order you can find inside my branches. And all this you can test at http://notecad.xyz