solvespace/solvespace #262: Constant CPU load

diggit, 2017-07-10:
### System information

SolveSpace version: 3.0~33b6e517 (It would be nice to have it selectable to copy)

Operating system: Archlinux

### Expected behavior

Minimal CPU load when not focused or focused and not doing anything.

### Actual behavior

Constant CPU load around 10% in average (~2s), even when unfocused and somewhere in background.

### Additional information

Processor i7-3610QM running above 3GHz, not exactly slow one...

whitequark, 2017-07-10:
Can you record a profile using `perf record solvespace` and then format it using `perf report | head -n 50`?

diggit, 2017-07-11:
Here it is:
```
# Total Lost Samples: 0
#
# Samples: 188K of event 'cycles:u'
# Event count (approx.): 105741425621
#
# Overhead  Command     Shared Object               Symbol                                                                                                                                                                                                                                                                                                                                                                                                                               
# ........  ..........  ..........................  .....................................................................................................................................................................................................................................................................................................................................................................................................................................
#
     2.35%  solvespace  libc-2.25.so                [.] _int_malloc
     2.34%  solvespace  solvespace                  [.] SolveSpace::TextWindow::Paint
     1.32%  solvespace  libc-2.25.so                [.] _int_free
     1.28%  solvespace  libpthread-2.25.so          [.] pthread_mutex_lock
     1.16%  solvespace  libpthread-2.25.so          [.] __pthread_mutex_unlock_usercnt
     1.15%  solvespace  solvespace                  [.] SolveSpace::RgbaColor::From
     0.87%  solvespace  libgobject-2.0.so.0.5200.3  [.] g_type_check_instance_is_a
     0.86%  solvespace  libc-2.25.so                [.] malloc
     0.83%  solvespace  libc-2.25.so                [.] __memcmp_sse4_1
     0.76%  solvespace  solvespace                  [.] SolveSpace::RgbaColor::FromFloat
     0.74%  solvespace  solvespace                  [.] SolveSpace::Vector3f::From
     0.64%  solvespace  solvespace                  [.] SolveSpace::Canvas::Fill::Equals
     0.61%  solvespace  solvespace                  [.] SolveSpace::BitmapFont::LocateGlyph
     0.54%  solvespace  solvespace                  [.] std::_Rb_tree<char32_t, std::pair<char32_t const, SolveSpace::BitmapFont::Glyph>, std::_Select1st<std::pair<char32_t const, SolveSpace::BitmapFont::Glyph> >, std::less<char32_t>, std::allocator<std::pair<char32_t const, SolveSpace::BitmapFont::Glyph> > >::_M_lower_bound
     0.51%  solvespace  libc-2.25.so                [.] free
     0.50%  solvespace  solvespace                  [.] SolveSpace::IdList<SolveSpace::Canvas::Fill, SolveSpace::Canvas::hFill>::FindByIdNoOops
     0.48%  solvespace  libc-2.25.so                [.] pthread_mutex_lock
     0.47%  solvespace  solvespace                  [.] SolveSpace::IdList<SolveSpace::OpenGl2Renderer::SMeshListItem, SolveSpace::Canvas::hFill>::FindByIdNoOops
     0.47%  solvespace  libGLX.so.0.0.0             [.] 0x0000000000006c8e
     0.47%  solvespace  libc-2.25.so                [.] __memmove_sse2_unaligned_erms
     0.45%  solvespace  solvespace                  [.] SolveSpace::SIndexedMesh::AddPixmap
     0.45%  solvespace  libglib-2.0.so.0.5200.3     [.] g_hash_table_lookup
     0.43%  solvespace  solvespace                  [.] SolveSpace::Canvas::GetFill
     0.43%  solvespace  solvespace                  [.] std::less<char32_t>::operator()
     0.41%  solvespace  libc-2.25.so                [.] __libc_calloc
     0.39%  solvespace  libc-2.25.so                [.] malloc_consolidate
     0.38%  solvespace  solvespace                  [.] std::__fill_n_a<unsigned int*, unsigned long, unsigned int>
     0.38%  solvespace  libX11.so.6.3.0             [.] XGetVisualInfo
     0.37%  solvespace  solvespace                  [.] std::__fill_n_a<SolveSpace::SIndexedMesh::Vertex*, unsigned long, SolveSpace::SIndexedMesh::Vertex>
     0.37%  solvespace  libGLX_mesa.so.0.0.0        [.] 0x000000000001a096
     0.31%  solvespace  solvespace                  [.] SolveSpace::RgbaColor::Equals
     0.30%  solvespace  libgobject-2.0.so.0.5200.3  [.] g_signal_emit_valist
     0.29%  solvespace  libpthread-2.25.so          [.] __pthread_rwlock_unlock
     0.29%  solvespace  libxcb.so.1.1.0             [.] xcb_send_request_with_fds64
     0.29%  solvespace  libc-2.25.so                [.] strlen
     0.27%  solvespace  solvespace                  [.] 0x00000000001c6572
     0.26%  solvespace  solvespace                  [.] std::vector<SolveSpace::SIndexedMesh::Vertex, std::allocator<SolveSpace::SIndexedMesh::Vertex> >::operator[]
     0.25%  solvespace  solvespace                  [.] std::vector<unsigned int, std::allocator<unsigned int> >::operator[]
     0.25%  solvespace  solvespace                  [.] std::vector<SolveSpace::SIndexedMesh::Vertex, std::allocator<SolveSpace::SIndexedMesh::Vertex> >::size
```
_Most of time was solvespace in background. Runnning 6.7 minutes._

whitequark, 2017-07-11:
That's interesting. Looks like something in your window system continuously requests the text window to be repainted. How's your desktop set up?

diggit, 2017-07-11:
I am using tiling wm i3. No compositing manager running. It's quite bare setup, nothing fancy.

I tried to run solvespace in fluxbox and if textwindow was not visible, CPU load dropped to 0. 
- switching to other virtual desktop worked
- minimizing text window worked too, but as it is not regular window, I couldn't find a way to restore it

Whenever it was visible, CPU load jumped back to 10%. 

If I switch to different virtual desktop in i3, there is no change and i3 has no feature like minimizing window (tiling WM...).

whitequark, 2017-07-11:
I use i3 too and don't observe such behavior. Weird.

diggit, 2017-07-11:
I have i3 built from git quite recently, comparing to last stable release from Nov 8, 2016 . Not sure which version you use.

Fluxbox test showed, that text window is being constantly repainted once it is visible. This is not ok IMO.

whitequark, 2017-07-11:
I'm not choosing when to repaint it while idle, GTK does. I'm not sure what's going on here.

whitequark, 2018-07-31:
I've done major restructuring of the platform backend recently, which might have fixed this. Can you please recheck?

diggit, 2018-08-26:
Issue seems solved, thanks!

closed 2018-08-26 by diggit