solvespace/solvespace #265: Set up crowdfunding stream

traverseda, 2017-07-15:
There should be a way for people to give the project money. I hear patreon is a good option.

whitequark, 2017-07-15:
This is not necessary. The limiting resource is not money (SolveSpace has one full-time programmer funded by M-Labs, though right now they are on hiatus) but time, specifically, time I spend on code review, merges and testing. I've had a lot of personal stuff going on recently and was unavailable to do much, which is why the project has stalled since April; that should change soon.

Also, quite frankly, most payout mechanisms suck if you're not a resident of US or EU. For the amount of donations a typical open-source project receives, it is a waste of time.

closed 2017-07-15 by whitequark

traverseda, 2017-07-15:
Well, I'd encourage you to diversify your funding streams and find a way to compensate third-party contributors. It's a generally useful thing to do.

BPLRFE, 2017-07-15:
Other things to think about:
- Please mention what you wrote above on the homepage. I am sure a lot of people would be willing to
pay via Paypal or similar (even i would consider a small donation) if it would help ...
- Please consider ways how people COULD contribute to the project
 > turning the homepage into a wiki (like OpenSCAD or FreeCAD) and giving some people editor
    rights so that the work on the homepage can be shared amongst more people
> mentioning writing code and doing a pull request on github
> mentioning doing written tutorials (and sharing/linking them on the homepage)
> mentioning doing video tutorials (and sharing/linking them on the homepage)
> think about giving one or two more people on github collaborator status being able
   to merge pull requests and closing issues ...
Since the program was put to GPL it would be great if the project would benefit from people
wanting to contribute.

whitequark, 2017-07-15:
The website already exists in a form that's fairly easy to update but the deployment isn't configured, which is a tad annoying.

Linking to 3rd party tutorials is an excellent idea, thanks.

I have not so far been satisfied with the code quality of external contributions. It is regretful but true that accepting pull requests ultimately often requires *more* of my time than implementing a feature or fixing a bug myself (or at least that was true in every instance so far, except perhaps localizations), and with my time being the limited resource here... you see the problem. I would like to put more effort into mentoring once I clear the backlog of code to be merged, which is the current priority.

BPLRFE, 2017-07-16:
So would you prefer creating issues for the Github/Solvespace/Website project or maybe you could
add a small readme there to describe how to add (after forking) something to the website and make a pull request ...

whitequark, 2017-07-16:
@BPLRFE Sure, issues are good. A bit later I'll properly document the workflow and we could do PRs.

Evil-Spirit, 2017-07-18:
@whitequark,
>I have not so far been satisfied with the code quality of external contributions. It is regretful but true that accepting pull requests ultimately often requires more of my time than implementing a feature or fixing a bug myself

This is not a reason for rejecting PRs. I think we can make a special branch (community branch) and merge all the good PRs and let people use it, test it and fix it themselves. When features will be proven, we can just merge it.
If we will act not in the way driven by our perfectionism, but in the way that allows other people do contribution, we can make SolveSpace more popular. BTW, if I will start to do as my perfectionism told me, I will just want to start the new SolveSpace from scratch. I am also think what I can implement new features from scratch faster than fixing already exising ugly code. But this is not the way how things in real life work. Project management is the ability to use low resources in more producitve way; I think what for management's goal is better to have team of 10 people even if it can be possible to hire one programmer who can be more productive than the 10. Ofc, this is not nessesary for the small projects, but better for the big in future.  So, I always have ambitions to do all by myself, but now I've start thinking about how I can manage the available resources to reach to goal. Anyway, the people must own the project, and must have way to contibute in clean and transparent way, where nor you nor anyone should be a bottleneck.

traverseda, 2017-07-18:
See also: [The cathedral and the bizarre](http://www.catb.org/esr/writings/cathedral-bazaar/cathedral-bazaar/). Translations available [here](http://www.catb.org/esr/writings/cathedral-bazaar/).

whitequark, 2017-07-18:
> This is not a reason for rejecting PRs. I think we can make a special branch (community branch) and merge all the good PRs and let people use it, test it and fix it themselves. When features will be proven, we can just merge it in master.

I have never seen this work in any open-source project. Why would it? Most people just want to scratch their own itch and don't have enough context to predict interactions with other code.

Open-source projects that merge a lot of external contributions have two traits in common:
  * They have an extensive battery of tests, and
  * They have a lot of people performing code review.
This still doesn't result in rapid merges. For example, even simple PRs for Rust take many weeks to land.

If we had tests that exercise most of SolveSpace's code in an automated way, this would be possible, but right now it's just a good way to have incomplete and buggy features.

I *am* working to bring SolveSpace to the point where contributing to it is easy, but we aren't there yet.

whitequark, 2017-07-18:
To add to this, I consider an application with few features that work very well far more useful than an application that has everything and a kitchen sink but little complete parts. Does FreeCAD have a working assembly workbench yet? Last time I looked they had three and none of them were usable. That's where your approach ends in.

SolveSpace is as usable as it is *precisely because* most of its parts were developed with deep attention to detail. It's hard to scale this beyond a single person; it's also necessary to, and I'm definitely going to do it.

Evil-Spirit, 2017-07-19:
> I have never seen this work in any open-source project. Why would it? Most people just want to scratch their own itch and don't have enough context to predict interactions with other code.

Yes, this is resonable argument. But how you can predict people intensions? May be this is the first commit of the person who will contribute in a quality way in future. If you are just rejecting PR, no PRs will be from this person in future. So, no chance to achieve quality contributors without approving thir first PR, no chance to have quality contrubutions in future without a bad ones before. The time is NOW, and people tring to contibute NOW, but we are not friendly for them. Accepting PRs is just saying "hello, people, we like your work".

whitequark, 2017-07-19:
> May be this is the first commit of the person who will contribute in a quality way in future. If you are just rejecting PR, no PRs will be from this person in future. So, no chance to achieve quality contributors without approving their first PR, no chance to have quality contrubutions in future without a bad ones before.

I absolutely agree that we need to accept PRs and help people become productive contributors. But I can only do so many things at a time, and so there will be compromises.

I've already worked out the CLA question with Jonathan, so the major obstacle to merging 3rd party code is cleared.

whitequark, 2017-07-20:
@BPLRFE I've written [instructions](https://github.com/solvespace/solvespace-web#solvespaces-website) for updating the SolveSpace website. Feel free to submit or work on [issues](https://github.com/solvespace/solvespace-web/issues).