solvespace/solvespace #266: Compilation error: Backtrace_LIBRARY (ADVANCED)

probonopd, 2017-07-20:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Backtrace_LIBRARY (ADVANCED)
    linked by target "solvespace-core" in directory solvespace/src
```

Full Travis CI build log: https://travis-ci.org/probonopd/solvespace/jobs/255811493#L986-L989

whitequark, 2017-07-21:
>  This job ran on our Trusty environment, which was updated on Wednesday, July 12th. Read all about it in the docs and take note that you can add group: deprecated-2017Q3 in your .travis.yml file to use the previous version.

Try this and check if it helps.

probonopd, 2017-07-21:
Unfortunately it did not help. https://github.com/probonopd/solvespace/blob/patch-1/.travis.yml

whitequark, 2017-07-21:
No idea sorry. On Linux, the `backtrace()` function is in glibc, so the check for it should not fail:

    -- Looking for backtrace
    -- Looking for backtrace - not found

You might want to enable CMake's verbose output to find out why.

whitequark, 2017-07-21:
Ah this reproduces on master. I'll look into it.

whitequark, 2018-07-31:
The backtrace library is no longer a mandatory dependency (it's been made optional), so I believe this is fixed.

closed 2018-07-31 by whitequark