solvespace/solvespace #267: Wrong detection of redundant constraint

alxv-su, 2017-07-21:
### System information

SolveSpace version: 2.3~

Operating system: Linux Mint Debian Edition 2 Betsy (using repos from Debian Jessie)

### Expected behavior

Program should solve sketch correctly. DOF should be equal to 0.

### Actual behavior

"SOLVE FAILED!" text appears in "Property Browser" window.

### Additional information

Problematic file attached: [bug-angle.slvs.zip](https://github.com/solvespace/solvespace/files/1165143/bug-angle.slvs.zip)

whitequark, 2017-07-21:
Confirmed in master. A workaround is to constrain the angle of `r005-line-segment` to the X axis, not the YZ reference plane.

phkahler, 2023-08-10:
This is not longer flagged as redundant, but the sketch still breaks easily unless we delete the angle constraint. It can be constrained against a horizontal line fine, but changing the angle value then breaks it (red).