solvespace/solvespace #270: CONTRIBUTING.md - How to build the project with Xcode and contribute from there.

ss0git, 2017-07-21:
### System information

SolveSpace version: any

Operating system: macOS Sierra
### Expected behavior

Since "Xcode command tools" are required to build anything on the mac, and those in turn require you to have Xcode it would make sense to use it to contribute to the project.

mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -G Xcode
Start Xcode and open the generated project file and build/debug the project from there.

### Actual behavior

mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make

whitequark, 2017-07-21:
Does this actually need -DCMAKE_BUILD_TYPE, or can you select that in XCode? I misplaced my macOS VM, can you please check?

ss0git, 2017-07-21:
Xcode has Build Schemes. And in each Build Scheme you get Build, Run, Test, Profile, Analyze and Archive. For each of these you can select Debug, Release, MinSizeRel and RelWithDebInfo as Build configurations.

That being said running "cmake .. -DCMAKE_BUILD_TYPE=Release -G Xcode" and "cmake .. -G Xcode" generate two somewhat different files with about 848 differences.

Attached generated folders for both 
[Archive.zip](https://github.com/solvespace/solvespace/files/1166371/Archive.zip)
solvespace.xcodeproj/project.pbxproj is different in each too.

whitequark, 2017-07-21:
I think what's going on here is the hashes and order of build schemes changed.

whitequark, 2017-07-21:
32120ea2

closed 2017-07-21 by whitequark

ss0git, 2017-07-21:
Folder size is not exactly the same either but in Xcode they seem to behave exactly the same.

whitequark, 2017-07-21:
I believe I disable benchmarks in Debug builds. This is not terribly important.