solvespace/solvespace #271: Running the Cmake generated code in Xcode ends in "Command /bin/sh failed with exit code 2"

ss0git, 2017-07-21:
### System information

SolveSpace version: any

Operating system: macOS

### Expected behavior

Project should build.

### Actual behavior

```
echo "Bundling executable solvespace-cli"
Bundling executable solvespace-cli
/usr/local/Cellar/cmake/3.9.0/bin/cmake -E copy /Users/me/Desktop/solvespace/build/bin/Debug/solvespace-cli /Users/me/Desktop/solvespace/build/bin/solvespace.app/Contents/MacOS
echo "Bundling shared library /usr/local/lib/libpng.dylib"
Bundling shared library /usr/local/lib/libpng.dylib
/usr/local/Cellar/cmake/3.9.0/bin/cmake -E copy /usr/local/lib/libpng.dylib /Users/me/Desktop/solvespace/build/bin/solvespace.app/Contents/MacOS/libpng.dylib
Error copying file "/usr/local/lib/libpng.dylib" to "/Users/me/Desktop/solvespace/build/bin/solvespace.app/Contents/MacOS/libpng.dylib".
make: *** [solvespace_buildpart_1] Error 1
Command /bin/sh failed with exit code 2
```

### Additional information

I think this one has something to do with the command below. It is supposed to copy solvespace-cli inside solvespace.app/Contents/MacOS/ (notice MacOS is a folder). This folder doesn't already exist like in the normal "make" scaffold and the copy command copies and renames solvespace-cli executable to MacOS and after that all other copy commands to that folder fail too.

Temporary fix: Create the folder yourself and Run it again. That should get you going. Don't know how to fix it though, maybe the Cmake generator should do something.

```
Bundling executable solvespace-cli
/usr/local/Cellar/cmake/3.9.0/bin/cmake -E copy /Users/me/Desktop/solvespace/build/bin/Debug/solvespace-cli /Users/me/Desktop/solvespace/build/bin/solvespace.app/Contents/MacOS
```

whitequark, 2017-07-21:
ah I think I messed up with parallelism or something, thanks

whitequark, 2017-07-21:
Does this help?
```patch
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 4363bcf0..dd8d22de 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -374,13 +374,10 @@ endif()
 # solvespace macOS package
 
 if(APPLE)
-    set(bundle     solvespace)
-    set(bundle_bin ${EXECUTABLE_OUTPUT_PATH}/${bundle}.app/Contents/MacOS)
+    set(bundle_bin ${EXECUTABLE_OUTPUT_PATH}/solvespace.app/Contents/MacOS)
 
-    add_custom_command(TARGET solvespace POST_BUILD
-        COMMAND ${CMAKE_COMMAND} -E copy $<TARGET_FILE:solvespace-cli> ${bundle_bin}
-        COMMENT "Bundling executable solvespace-cli"
-        VERBATIM)
+    list(APPEND bundle_cmds
+        COMMAND ${CMAKE_COMMAND} -E copy $<TARGET_FILE:solvespace-cli> ${bundle_bin})
 
     foreach(lib ${platform_BUNDLED_LIBS})
         get_filename_component(name ${lib} NAME)
@@ -388,23 +385,26 @@ if(APPLE)
         execute_process(COMMAND otool -D ${lib}
             OUTPUT_VARIABLE canonical_lib OUTPUT_STRIP_TRAILING_WHITESPACE)
         string(REGEX REPLACE "^.+:\n" "" canonical_lib ${canonical_lib})
-        add_custom_command(TARGET ${bundle} POST_BUILD
+        list(APPEND bundle_cmds
             COMMAND ${CMAKE_COMMAND} -E copy ${lib} ${bundle_bin}/${name}
             COMMAND install_name_tool -change ${canonical_lib} @executable_path/${name}
-                    $<TARGET_FILE:${bundle}>
+                    $<TARGET_FILE:solvespace>
             COMMAND install_name_tool -change ${canonical_lib} @executable_path/${name}
-                    $<TARGET_FILE:solvespace-cli>
-            COMMENT "Bundling shared library ${lib}"
-            VERBATIM)
+                    $<TARGET_FILE:solvespace-cli>)
     endforeach()
 
-    add_custom_command(OUTPUT ${EXECUTABLE_OUTPUT_PATH}/${bundle}.dmg
-        COMMAND ${CMAKE_COMMAND} -E remove ${EXECUTABLE_OUTPUT_PATH}/${bundle}.dmg
-        COMMAND hdiutil create -srcfolder ${EXECUTABLE_OUTPUT_PATH}/${bundle}.app
-                ${EXECUTABLE_OUTPUT_PATH}/${bundle}.dmg
-        DEPENDS $<TARGET_FILE:${bundle}>
-        COMMENT "Building ${bundle}.dmg"
+    add_custom_command(TARGET ${solvespace} POST_BUILD
+        ${bundle_cmds}
+        COMMENT "Building solvespace.app"
+        VERBATIM)
+
+    add_custom_command(OUTPUT ${EXECUTABLE_OUTPUT_PATH}/solvespace.dmg
+        COMMAND ${CMAKE_COMMAND} -E remove ${EXECUTABLE_OUTPUT_PATH}/solvespace.dmg
+        COMMAND hdiutil create -srcfolder ${EXECUTABLE_OUTPUT_PATH}/solvespace.app
+                ${EXECUTABLE_OUTPUT_PATH}/solvespace.dmg
+        DEPENDS $<TARGET_FILE:solvespace>
+        COMMENT "Building solvespace.dmg"
         VERBATIM)
-    add_custom_target(${bundle}-dmg ALL
-        DEPENDS ${EXECUTABLE_OUTPUT_PATH}/${bundle}.dmg)
+    add_custom_target(solvespace-dmg ALL
+        DEPENDS ${EXECUTABLE_OUTPUT_PATH}/solvespace.dmg)
 endif()
```

ss0git, 2017-07-21:
It doesn't, MacOS folder is still not there to copy to. I'm new to Cmake but your initial code seems fine.

This line is ok:
+    set(bundle_bin ${EXECUTABLE_OUTPUT_PATH}/solvespace.app/Contents/MacOS)
but the folder MacOS doesn't exist. From our earlier discussion it would seem:

` cmake .. -DCMAKE_BUILD_TYPE=Release` creates the directory
` cmake .. -DCMAKE_BUILD_TYPE=Release -G Xcode` doesn't create the directory
` cmake .. -G Xcode` doesn't create the directory

I think at line 380 we could add a line to create the directory, until a more elegant solution appears.
from:
```
    add_custom_command(TARGET solvespace POST_BUILD
        COMMAND ${CMAKE_COMMAND} -E copy $<TARGET_FILE:solvespace-cli> ${bundle_bin}
        COMMENT "Bundling executable solvespace-cli"
VERBATIM)
```
to:
```
    add_custom_command(TARGET solvespace POST_BUILD
        COMMAND ${CMAKE_COMMAND} -E make_directory ${bundle_bin}
        COMMAND ${CMAKE_COMMAND} -E copy $<TARGET_FILE:solvespace-cli> ${bundle_bin}
        COMMENT "Bundling executable solvespace-cli"
VERBATIM)
```

whitequark, 2017-07-21:
Can you confirm that with this change, the project can be built from scratch reliably?

ss0git, 2017-07-21:
It fixes the folder issue. That being said it crashes on launch for other reasons.

Should I create a new issue or continue here? Crashes on

```
//-----------------------------------------------------------------------------
// Discovery and loading of our resources (icons, fonts, templates, etc).
//
// Copyright 2016 whitequark
//-----------------------------------------------------------------------------
#include <zlib.h>
#include <png.h>
#include <regex>
#include "solvespace.h"

namespace SolveSpace {

//-----------------------------------------------------------------------------
// Resource loading functions
//-----------------------------------------------------------------------------

std::string LoadString(const std::string &name) {
    size_t size;
    const void *data = Platform::LoadResource(name, &size);
    ...
```
then in platform.cpp
```
const void *LoadResource(const std::string &name, size_t *size) {
    static std::map<std::string, std::string> cache;

    auto it = cache.find(name);
    if(it == cache.end()) {
        ssassert(ReadFile(ResourcePath(name), &cache[name]), "Cannot read resource");
        ...
```

The value of the variable name = "locales.txt"

```
en
File /Users/ieorjiwerjl/Desktop/solvespace/src/platform/platform.cpp, line 560, function LoadResource:
Assertion 'Cannot read resource' failed: ((ReadFile(ResourcePath(name), &cache[name])) == false).
Backtrace:
 0: 0   solvespace                          0x0000000100194b48 _ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_ + 120
 1: 1   solvespace                          0x000000010018e4e3 _ZN10SolveSpace8Platform12LoadResourceERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEPm + 835
 2: 2   solvespace                          0x00000001001b8efc _ZN10SolveSpace10LoadStringERKNSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEE + 60
 3: 3   solvespace                          0x00000001001cc603 _ZN10SolveSpace7LocalesEv + 275
 4: 4   solvespace                          0x00000001001ce2f7 _ZN10SolveSpace9SetLocaleIZNS_9SetLocaleERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEE3$_1EEbT_ + 23
 5: 5   solvespace                          0x00000001001ce2cd _ZN10SolveSpace9SetLocaleERKNSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEE + 29
 6: 6   solvespace                          0x00000001000123e8 main + 1000
 7: 7   libdyld.dylib                       0x00007fff904755c9 start + 1
 8: 8   ???                                 0x0000000000000001 0x0 + 1
(lldb) 
```

whitequark, 2017-07-23:
49c64f7f

closed 2017-07-23 by whitequark