solvespace/solvespace #272: Xcode building the project missing Info.plist file in solvespace app folder?

ss0git, 2017-07-21:
That being said it crashes on launch for other reasons. There are two issues I can spot right now, and I think they are related.

- In the solvespace.app folder there are two folders MacOS and Resources. A file that is missing and should be there is info.plist, it's there in the 'make' build and macOS apps in general have it.

- It seems to fail to find its resources? The file is there in the Resources folder but a missing Info.plist file means the app doesn't have the values it needs to start. Crashes on line:

```
//-----------------------------------------------------------------------------
// Discovery and loading of our resources (icons, fonts, templates, etc).
//
// Copyright 2016 whitequark
//-----------------------------------------------------------------------------
#include <zlib.h>
#include <png.h>
#include <regex>
#include "solvespace.h"

namespace SolveSpace {

//-----------------------------------------------------------------------------
// Resource loading functions
//-----------------------------------------------------------------------------

std::string LoadString(const std::string &name) {
    size_t size;
    const void *data = Platform::LoadResource(name, &size);
    ...
```
and in platform.cpp
```
const void *LoadResource(const std::string &name, size_t *size) {
    static std::map<std::string, std::string> cache;

    auto it = cache.find(name);
    if(it == cache.end()) {
        ssassert(ReadFile(ResourcePath(name), &cache[name]), "Cannot read resource");
        ...
```

The value of the variable name = "locales.txt"

```
en
File /Users/me/Desktop/solvespace/src/platform/platform.cpp, line 560, function LoadResource:
Assertion 'Cannot read resource' failed: ((ReadFile(ResourcePath(name), &cache[name])) == false).
Backtrace:
 0: 0   solvespace                          0x0000000100194b48 _ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_ + 120
 1: 1   solvespace                          0x000000010018e4e3 _ZN10SolveSpace8Platform12LoadResourceERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEPm + 835
 2: 2   solvespace                          0x00000001001b8efc _ZN10SolveSpace10LoadStringERKNSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEE + 60
 3: 3   solvespace                          0x00000001001cc603 _ZN10SolveSpace7LocalesEv + 275
 4: 4   solvespace                          0x00000001001ce2f7 _ZN10SolveSpace9SetLocaleIZNS_9SetLocaleERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEE3$_1EEbT_ + 23
 5: 5   solvespace                          0x00000001001ce2cd _ZN10SolveSpace9SetLocaleERKNSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEE + 29
 6: 6   solvespace                          0x00000001000123e8 main + 1000
 7: 7   libdyld.dylib                       0x00007fff904755c9 start + 1
 8: 8   ???                                 0x0000000000000001 0x0 + 1
(lldb) 
```

whitequark, 2017-07-23:
I think your analysis is correct, here's how it's locating the path: https://github.com/solvespace/solvespace/blob/master/src/platform/platform.cpp#L475-L500

Not sure about Info.plist file.

whitequark, 2017-07-23:
I have a [MacOSXBundleInfo.plist.in file](https://github.com/solvespace/solvespace/blob/master/cmake/MacOSXBundleInfo.plist.in), which [should](https://cmake.org/cmake/help/v3.7/prop_tgt/MACOSX_BUNDLE_INFO_PLIST.html) be used for the target that [specifies MACOSX_BUNDLE](https://github.com/solvespace/solvespace/blob/master/src/CMakeLists.txt#L302-L305).

whitequark, 2017-07-23:
Also wow, I had no idea [BundleUtilities](https://cmake.org/cmake/help/v3.9/module/BundleUtilities.html) existed, I did not need to mess with `install_name_tool` manually after all. Thanks!

whitequark, 2018-07-18:
The `Info.plist` is generated by cmake command and not regenerated if you manually remove `SolveSpace.app` and call `make`. As for BundleUtilities, master simply links everything statically now.

closed 2018-07-18 by whitequark