solvespace/solvespace #274: Wrong detection of redundant constraint #2

alxv-su, 2017-07-25:
### System information

SolveSpace version: 2.3~

Operating system: Linux Mint Debian Edition 2 Betsy (using repos from Debian Jessie)

### Expected behavior

Program should solve sketch correctly with any distance value.

### Actual behavior

Solve failed if distance greater than 2645 (see screenshots).

### Additional information

Screenshots:
[image]
[image]
Problematic files: [bug-distance.zip](https://github.com/solvespace/solvespace/files/1173462/bug-distance.zip)

whitequark, 2017-07-25:
This is already fixed in the `master` branch.

whitequark, 2017-07-25:
(Even if it wasn't, integrating the code in #225 would fix bugs of this sort as well.)

closed 2017-07-25 by whitequark