solvespace/solvespace #275: Add  in README info about OpenGL flags for compilling

Symbian9, 2017-07-26:
### System information

SolveSpace version: 'master'

### Additional info

* https://github.com/solvespace/solvespace/issues/17#issuecomment-317885325

whitequark, 2017-07-26:
Sorry, no. `-DOPENGL` only works on Windows, results in a sub-par experience, and will not exist for a long time anyway.

closed 2017-07-26 by whitequark

Symbian9, 2017-07-26:
Also in "Releases" section names of binary must include info for what OpenGL version it compiled

* for OpenGL 1.x: `solvespace-2.3_gl1.exe`

* for OpenGL|ES 2.x - `solvespace-2.3_gl2.exe`

whitequark, 2017-07-26:
There is no need to ever have a GL1-only Windows executable because ANGLE takes care of translating our OpenGL 3/ES 2 calls into corresponding D3D9/D3D11 calls.

Symbian9, 2017-07-26:
Many user need OpenGL1.x only binary for older PC.

Also for building binary for Linux this deferentiation also needed.

whitequark, 2017-07-26:
> Many user need OpenGL1.x only binary for older PC.

They can keep using SolveSpace 2.3.

> Also for building binary for Linux this deferentiation also needed.

This is not supported now and will never be supported again, since it's incompatible with using GTK 3.