solvespace/solvespace #276: Not buildable out-of-box on Linux

helifino, 2017-07-27:
### System information

SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...)
2.3 release

Operating system: (e.g. Debian testing)
Linux Mint 18.1 Serena 64-bit

### Expected behavior

What should have happened?
Successful build

### Actual behavior

What actually happened?
Several errors.
 

1. No pre-built .deb package as listed in the README
2. No mention of the required compiler or the presence of 'git' in the dependencies portion of the README
3. git submodule dependency retrieval not satisfied because user doesn't have knowledge of git
4. cmake also assumes user downloaded source via git, throws lack-of-git related errors

### Additional information

whitequark, 2017-07-27:
Note that the 2.3 release is in Debian repositories; you do not need to build it from source.

helifino, 2017-07-27:
There is no indication of this in the documentation- website or README

Symbian9, 2017-07-27:
Similar related to known PPA

* https://github.com/solvespace/solvespace/issues/142

There are AppImage's produced by @probonopd, but they not follow release versions and mostly look like compiled from master

* https://github.com/solvespace/solvespace/issues/17

Symbian9, 2017-08-01:
Latest discussion on forum
* http://solvespace.com/forum?action=viewthread&parent=2042&tt=1501555666

Symbian9, 2017-08-01:
@helifino, try compile under xUbuntu 16.04 LTS (LinuxMint 18.1) using this
```
sudo apt install cmake zlib1g-dev libpng-dev libcairo2-dev libfreetype6-dev sudo apt install libjson-c-dev libfontconfig1-dev libgtkmm-3.0-dev libpangomm-1.4-dev libgl-dev libglu-dev libspnav-dev
git clone -b ranged_constraints https://github.com/Evil-Spirit/solvespace-master.git
cd solvespace-master
git submodule update --init extlib/libdxfrw
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make
sudo make install
```

helifino, 2017-08-02:
None of the above is in the README.md, and the package install line doesn't include a compiler.

whitequark, 2017-08-02:
@helifino Sure, you're right. I'll update the README when I have a bit of time.

Symbian9, 2017-08-07:
> and the package install line doesn't include a compiler

@foror, could you fix it?

foror, 2017-08-08:
I'm not an expert in this topic, but probably there needs to add gcc: sudo apt install gcc cmake zlib1g-dev libpng-dev libcairo2-dev libfreetype6-dev sudo apt install libjson-c-dev libfontconfig1-dev libgtkmm-3.0-dev libpangomm-1.4-dev libgl-dev libglu-dev libspnav-dev

whitequark, 2017-08-08:
Fixed in afe2279a.

nick87720z, 2017-09-05:
Hello. Though this issue is already closed, i hope - this still may be read.
Some parts may affect entire release system rather, than separate releases.

Not so long ago i completed working gentoo ebuild for solvespace 2.3 and 9999. While for 9999 it is dumb and simple (no need for source patching / modification before configure/build), version 2.3 required following steps:

1. Before to configure, to prevent some git-related errors:
```
 sed -ie '/^include(GetGitCommitHash)/s/^/# /`
          /set(GIT_COMMIT_HASH/s/^# //' ./CMakeLists.txt`
```
2. This release is not so completed, as submodules dirs are empty. So, solution was to download some fixed vcs revisions for submodules, which are assumed to work for this release, then manually create expected dirs and unpack archives into them, from these urls:
```
angle_commit="29f908bf3261b8b1e2faf3454b10b811415bd26a"
cairo_commit="747cab741cf63f30aa6bc2b787fc13e7e7a2b859"
libdxfrw_commit="6f362317bf3f176b613be48512a88b78125c79f4"
libpng_commit="1.6.26"`
pixman_commit="7c6066b700c7cdd4aeb8be426b14b3a5f0de4b6c"
zlib_commit="1.2.11"
    https://github.com/solvespace/angle/archive/${angle_commit}.zip
    https://github.com/solvespace/cairo/archive/${cairo_commit}.zip
    http://git.savannah.gnu.org/cgit/freetype/freetype2.git/snapshot/freetype2-VER-2-7-1.tar.gz
    https://github.com/solvespace/libdxfrw/archive/${libdxfrw_commit}.zip
    https://github.com/glennrp/libpng/archive/v${libpng_commit}.tar.gz
    https://github.com/solvespace/pixman/archive/${pixman_commit}.zip
    https://github.com/madler/zlib/archive/v${zlib_commit}.tar.gz` 
```
Would be great if releases include submodules, thus not requiring their fetch manually.
Or... for fallback, it could just use system libs.

whitequark, 2017-09-05:
> This release is not so completed, as submodules dirs are empty. So, solution was to download some fixed vcs revisions for submodules, which are assumed to work for this release, then manually create expected dirs and unpack archives into them, from these urls:

You should have downloaded the "solvespace-source.tar.gz", not "Source code (tar.gz)" from the [releases page](https://github.com/solvespace/solvespace/releases). Unfortunately GitHub neither archives submodules in the latter link, nor permits me to remove it.

Also, there is no need for any of submodules except libdxfrw when building on Linux.

whitequark, 2017-09-05:
> Before to configure, to prevent some git-related errors:
>  sed -ie '/^include(GetGitCommitHash)/s/^/# /`
>           /set(GIT_COMMIT_HASH/s/^# //' ./CMakeLists.txt`

Regarding this, it is actually a problem with the published sources and I will fix it.

nick87720z, 2017-09-05:
Did not know. Tried solvespace-sources, its all ok, appart still requiring above sed command to be working.
First time i got existing ebuild, and probably, there was such link, but then i changed it, being unaware.

Symbian9, 2017-09-14:
Here are few working solutions and builds of 2.3 release for Fedora 24 & Ubuntu 16.04
* http://solvespace.com/forum.pl?action=viewthread&parent=1419&tt=1505280021

whitequark, 2018-07-31:
I've updated the released sources to contain embedded git hashes.

closed 2018-07-31 by whitequark

tbart, 2020-06-02:
@nick87720z: Care to share your 9999 ebuild? Probably push it to bugs.gentoo.org so other people can find it as well. Thanks in advance!

nick87720z, 2020-06-02:
@tbart Yeah, I have dozens ebuilds still not published. It's usually long process. Still can't get to create mature personal overlay.
https://bugs.gentoo.org/726806

StefanBruens, 2023-03-17:
Can someone please reupload a fixed source tarball for 3.1? The git hash is not set there (https://github.com/solvespace/solvespace/blob/b5333608e9d4826fcb6dc497a9e94e162267a84c/.github/workflows/source-tarball.yml is post-release).