solvespace/solvespace #278: ANGLE is slow

whitequark, 2017-08-02:
@Evil-Spirit [says](https://github.com/solvespace/solvespace/issues/17#issuecomment-319550418) that ANGLE is too slow and we should provide native OpenGL Windows builds. Questions:

 * Can we mitigate the ANGLE startup delay (which is significant)?
 * How much slowdown does ANGLE incur after startup? It's certainly fast enough for Chrome, so I would think it should not matter for us; we have less intense demands.
 * Can we easily determine if we should use ANGLE or OpenGL during startup, without dragging in too much support infrastructure?

Evil-Spirit, 2017-08-02:
@whitequark, I have tried to figure out what's happening and get the following information:
* probably, ANGLE mode DYAMIC_DRAW is slower when STATIC_DRAW even if we create and delete buffer immediately
* probably, the function D3D11CreateDeviceAndSwapChain called every frame and this is slows down (probably we are doing something wrong or misusing ANGLE in some way)

This is about things we draw in fake-immediate mode (meshes for UI, constraints edges, etc). All what we are drawing in batch mode is fast and ok.

whitequark, 2017-08-02:
@Evil-Spirit Idea: we could use EGL_ANGLE_platform_angle_opengl extension to request ANGLE to just forward our calls to OpenGL.

Evil-Spirit, 2017-08-02:
@whitequark, Yes if this will work in the right way, this would be great. If this firstly checks for native support of OpenGL > 1, and if it's not presented, uses ANGLE in DirectX mode.

whitequark, 2019-05-23:
@Evil-Spirit Yeah, that worked.

Fixed in e9b9dca.

closed 2019-05-23 by whitequark