solvespace/solvespace #279: Import image as 2D workplane background

EddieRich, 2017-08-02:
This is an enhancement similar to #143 except I would like to import an image.
There are 2D plans available on the web for model builders and it would be nice to have a background image on the 2D workplane that can be scaled and moved. Then the parts could be sketched on top of the image. Besides scale and move, the image should also have a transparency parameter.
Each specific workplane should have it's own background image.

Symbian9, 2017-08-02:
This feature implemented in SolveSpace 3.x branch, but not in SolveSpace 2.x.

@Evil-Spirit, is it possible backport this feature in 2.x?

EddieRich, 2017-08-02:
>

EddieRich, 2017-08-02:
whitequark, there is no sketch->Image option.
I suppose if an image could be added just like a line or circle, that would work.

whitequark, 2017-08-02:
It's not yet present in any released version, but yes, that's essentially how it works. I'm now closing this as duplicate, since you say that's fine.

closed 2017-08-02 by whitequark

Symbian9, 2017-08-02:
> I'm now closing this as duplicate, since you say that's fine.

He not say "that's fine", lol.

EddieRich, 2017-08-03:
Closing is fine. I just discovered the background bitmap image option (which is under line styles for some reason). That works as long as nothing needs to be constrained to the origin.

whitequark, 2017-08-03:
That's a less powerful version of the same feature, available in version 2.3, yes.