solvespace/solvespace #280: Problem with STL export & Windows/linux versions .slvs portability

nekromant, 2017-08-05:
### System information

SolveSpace version: windows: 2.3~7c1ca460

Operating system: Windows, Debian

### Expected behavior

Solvespace should export STL with no problems. Windows version should open files created in linux version with no problems. 

### Actual behavior

A certain more or less complex design created with linux version solvespace (from debian stretch repository) can't be exported by the linux version of solvespace (freezes with 100% CPU load during STL export). The linux version opens the file instantly though. The windows version can't even open the file: Freezes with "generating group 22/34". I've tried with both nvidia and intel graphics on an optimus-capable laptop. 

### Additional information

I have attached the problematic STL file to this bug report. 
[phone-box.zip](https://github.com/solvespace/solvespace/files/1202191/phone-box.zip)

whitequark, 2017-08-05:
I strongly suspect you're hitting some pathological numeric cases we fixed in baf9dc0aae573f017b8477e5d438665e90e2d5ae, and it differs between Linux and Windows because of different default rounding modes. Do you think you can use a nightly Windows build or make a Linux build yourself, and verify that it still crashes on master? It seems like I can't reproduce it locally.

nekromant, 2017-08-05:
@whitequark Thanks a lot for the tip, it worked in windows nightly build like a charm. I'll compile and see if it works in linux as well and post the results. And I do owe you a coffee ;)

nekromant, 2017-08-05:
@whitequark Yes, everything works on 49c64f7f1520160ad4584ca55382055f3111d4ac like a charm in linux as well. Thanks again for the assistance, I'm closing the issue.

closed 2017-08-05 by nekromant