solvespace/solvespace #283: Off-topic: Jacobian as a "general method" for underconstrained equation systems

dumblob, 2017-08-13:
Few months ago a discussion started about how to cope with underconstrained equation systems when it comes to "letting the user know it's most probably underconstrained" and "coping with underconstrained system in a more or less human-unsurprising manner". You can read the discussion on https://github.com/starwing/amoeba/issues/6 .

Do you think the Jacobian matrix and Jacobian as they're used in Solvespace could be useful in a similar ways also in the case of Amoeba?

whitequark, 2017-08-13:
@Evil-Spirit @jwesthues What do you think?

jwesthues, 2017-08-13:
I don't see much analogy between your problem and SolveSpace's. We write a Jacobian in order to (locally) transform the nonlinear constraint equations into a linear problem, but if I understand correctly you're dealing only with linear constraints to start with.

SolveSpace tries to avoid surprising behavior by solving underconstrained problems with a solution as close as possible (by various heuristics) to the solver's initial guess. I again don't see the analogy, since I don't think you have or otherwise need that guess.

Evil-Spirit, 2017-08-14:
@dumblob 
You should take a look here: https://github.com/solvespace/solvespace/blob/master/src/system.cpp#L561
Probably, this function can say to you which variables are "free" regardless of type of eq system you have : linear or non linear. For linear systems, the Jacobian consists of just constants, so you can easily construct it. 
The second what you need is rank test function.

dumblob, 2017-08-14:
>but if I understand correctly you're dealing only with linear constraints to start with.

That's correct.

@Evil-Spirit thank you for the pointers - you've kicked my thinking out of a local extreme! I'll try to give them a thought.

What do you think @starwing @hfossli ?

hfossli, 2017-08-14:
It was a nice-to-have feature which I never got around to look at closely enough. Thanks for the tip. I'm not familiar with this algorithm, but I think I found a viable solution. Would love to know how Ken Ferry and friends solved this in Auto Layout and Interface Builder (apple tools).

whitequark, 2017-08-19:
Looks like everyone got the answers they wanted, so, closing.

closed 2017-08-19 by whitequark