solvespace/solvespace #285: Change "split curves at intersection" so it works on text

DanielO, 2017-08-30:
### System information

SolveSpace version: 2.3
Operating system: MacOS Sierra 10.12.6

### Expected behavior
It would be nice to be able to "split curves at intersection" on text so I can merge it with another element in the same work plane.

See attached image for what I was creating - in the end I put the circle and text in 2 different work planes and extruded them independently but that was a bit of a pain.

[image]

whitequark, 2017-08-30:
I'm not sure how feasible is that with our current implementation. Maybe if we add a separate "turn text to entities" option?

DanielO, 2017-08-30:
Yes that would be great, I actually looked for such a thing as a work around! :)

whitequark, 2017-08-30:
As a very crude workaround you could export text as DXF and then re-import it back.

DanielO, 2017-08-30:
I tried that but the result was not good - maybe I did something wrong though - see the attached file.

I created the text, did Export 2d view as DXF, move original text, import DXF and that was the result.

whitequark, 2017-08-30:
There's no attached file...

DanielO, 2017-08-30:
Ah oops, I ignored the "we don't support that file type" error..

[issue-285.zip](https://github.com/solvespace/solvespace/files/1262493/issue-285.zip)

whitequark, 2017-08-31:
Ah, I see. This seems to be related to Bezier curves:
[image]

@Evil-Spirit do you have any quick ideas?

whitequark, 2017-08-31:
The DXF file seems OK to me so it must be the import feature at fault.

Evil-Spirit, 2017-09-01:
@whitequark 
Hm. this is my fault. I actually didn't implement it.
Look here
https://github.com/solvespace/solvespace/blob/master/src/importdxf.cpp#L671

Evil-Spirit, 2017-09-01:
@whitequark
The reason is what we are exported it like beziers degree of 2, but we can import only 3.

Evil-Spirit, 2017-09-01:
@jwesthues, what do you think about it?

whitequark, 2017-09-01:
@Evil-Spirit Right, that should have at least had `importer.unknownEntities++;`

jwesthues, 2017-09-02:
The curves from a TTF can be represented exactly as SolveSpace's cubic nonrational Bezier segments. Depending on the font, you may get an impractically large number of segments.

BPLRFE, 2017-09-02:
What about support of Sticky Fonts (CHR) ?
Since they can be converted to G-Code (See NCPlot Stickfont Program http://ncplot.com/stickfont/stickfont.htm) that seems to be a solution ?

whitequark, 2017-09-02:
@BPLRFE TTF stick fonts (e.g. [CamBam ones](http://mrrace.com/CamBam_Fonts/)) are already supported by SolveSpace:
[image]

Of course, because they are infinitely thin, the resulting entity contour cannot be extruded. (Neither could be CHR stick fonts if we had support for them). But you can still use them as construction entities, e.g. for documentation or reference, if you want. Also, if exporting as DXF, overlapping segments would be removed, and the resulting file could be used for engraving.

Evil-Spirit, 2017-09-03:
@jwesthues I mean, we can't repersent it using the current bezier curve entity. It doesn't support to be degree of 2, and can't be rational (support circles). Or you mean, TTF don't support circles?

jwesthues, 2017-09-03:
TTF uses non-rational quadratic Bezier pieces, exactly representable as SolveSpace's cubics.

Evil-Spirit, 2017-09-04:
Yes, this will be enough for ttf but probably not for general dxf

phkahler, 2023-11-20:
Isn't this is going to require converting a Text entity into spline entities. There is no way to represent a portion of a "text entity" which would be required to split them.

jwesthues, 2023-11-21:
@phkahler Yeah. TTF uses quadratic nonrational curves, but all quadratics are also exactly representable as our usual cubics. The number of segments per character depends on the font; it could be just ten, or it could be hundreds.