solvespace/solvespace #287: Regenerate full assembly tree (including assembled assembly)

Symbian9, 2017-09-06:
### System information

SolveSpace version: all

Operating system: all

### Expected behavior

Recursively regenerate assembly "B" linked in other assembly "C" as part when click `Space` button.

### Actual behavior

Currently if simple model "A", linked in assembly "B", was changed and saved separatly then this changes not going directly in assembly "C".

Details
* http://solvespace.com/forum.pl?action=viewthread&parent=2103&tt=1504574471

### Additional information

...

whitequark, 2017-09-06:
Depends on #78. Don't expect this any time soon.

Symbian9, 2017-09-06:
Cast @Evil-Spirit, @jwesthues

whitequark, 2017-09-06:
There is nothing they can do here, it's a fundamental limitation of the way the existing codebase is implemented.