solvespace/solvespace #291: program crash on any export except Export Image

sraynesford, 2017-09-14:
### System information

SolveSpace version: 2.3~7c1ca460

Operating system: Windows 10

### Expected behavior
Design exported as an stl file, or step file or as a graphic in a pdf file.

### Actual behavior
solvespace.exe runs at high cpu rate for a while.
"(Not Responding)" shows on title bar if app window is clicked while running.
Then Windows pop-up announces "solvespace.exe has stopped working".

### Additional information
In the case of step or pdf output, there is a zero-byte file.  
For the stl option, no output file is created.
Export as .png finishes correctly.

Design is intended for 3d printing.
Design starts with a sketch, lathed, then differenced with an extrusion to keep only a portion of the lathed object.
Full-circle lathed object exported as an stl, but was too large to print.

[fan motor air deflector.zip](https://github.com/solvespace/solvespace/files/1301723/fan.motor.air.deflector.zip)

sraynesford, 2017-09-14:
I deleted the extrusion and created a new extrusion of the same form starting on a workplane below the lathed part and then extruded to a plane above the part.  This exported without any further issue.

[fan motor air deflector 3.zip](https://github.com/solvespace/solvespace/files/1303152/fan.motor.air.deflector.3.zip)

whitequark, 2018-07-12:
I can confirm this is present even with the fix for #259.

whitequark, 2018-07-12:
@Evil-Spirit Looks like the export is spending a lot of time in `SEdgeList::CullExtraneousEdges`, any ideas?

Evil-Spirit, 2018-07-12:
May be BSP tree overflows stack when performed triangle booleans or hidden line remove?

gnydick, 2019-10-25:
i'm guessing this is getting me too, I created a simple file then closed, and now it hangs at open with "Not Responding". Is there a tmp file somewhere or where is the last-open-file saved? I didn't even save my work.

whitequark, 2019-10-25:
> Is there a tmp file somewhere or where is the last-open-file saved? I didn't even save my work.

See `configuration` → `autosave interval`. The next time you open the same file, you'll get prompted for whether you want to load the autosave.