solvespace/solvespace #292: Some question about the solver and *.slvs format

KmolYuan, 2017-09-14:
Hello. I made a Python 3 wrapper for the solver,

+ https://github.com/KmolYuan/python-solvespace

This project is using newest source code and compile without CMake. It can made `slvs.h` be import then used simply in a Python script (shown as [PyDemo](https://github.com/KmolYuan/python-solvespace/blob/master/PyDemo.py)).

If allowed, I would like to add it to the "solver" branch.

On the other hand, I want external programs can read and write *.slvs format, with fully Link / Assemble support, then solvespace files can be generated from C or Python.

Best regard.

Symbian9, 2017-09-22:
For the record, soon it could be used for implement SolveSpace solver in Blender 2.7x or later
* https://github.com/KmolYuan/python-solvespace/issues/2

whitequark, 2017-09-30:
@KmolYuan The "solver" branch exists because of copyright reasons, not for technical ones. If you are aiming for integrating your Python bindings, that should be done in the "master" branch, where active development happens. Are you fine with that?

KmolYuan, 2017-09-30:
Sure!

If it can be integrated into the CMake build process it would be better.

-----

By the way, the file [file.cpp] is using information from SK ([Sketch class]) of [SolveSpaceUI class], It seems to be unrelated to solver parts.

Is possible to separate SK and UI parts, or just using SolveSpaceUI directly to write a *.slvs format?

[file.cpp]: https://github.com/solvespace/solvespace/blob/master/src/file.cpp
[Sketch class]: https://github.com/solvespace/solvespace/blob/master/src/solvespace.h#L562
[SolveSpaceUI class]: https://github.com/solvespace/solvespace/blob/master/src/solvespace.h#L591

whitequark, 2017-09-30:
OK. I will ask you to sign a CLA very soon and then I will be able to integrate your code. I will handle CMake and basic testing for you.

Right now the solver that is exposed via the external interface can only solve for one group, and files have many. So, the exposed solver cannot interact with files. I'm not sure what's the best way to fix this, I've not really considered the binding evolution much.

KmolYuan, 2017-09-30:
For the file format section, maybe there should be a new external interface with the internal data format.

Because I want to put the results of the external design process into Solvespace. There would be used as auto-assemble function.

whitequark, 2017-09-30:
> For the file format section, maybe there should be a new external interface with the internal data format.

Unfortunately the way code is currently structured, the internal C++ interface does not check many invariants, so actually exposing it as-is in Python would cause a hard to debug crash in case of misuse. Of course we want to fix that but it is by no means quick.

The UI is not scriptable either.

whitequark, 2018-07-12:
Thank you for your contribution, and welcome. The SolveSpace project has a Contributor License Agreement; in order for me to be able to merge your changes, please sign it at https://cla-assistant.io/solvespace/solvespace. It will take only a small amount of your time.

whitequark, 2019-05-23:
@KmolYuan Are you still interested in integrating this?

KmolYuan, 2019-05-24:
[Python-Solvespace](https://github.com/KmolYuan/python-solvespace) project has been updated a new wrapper mechanism to [Cython](https://github.com/cython/cython) instead of SWIG at April 2019.

The old SWIG wrapper is terminated at https://github.com/KmolYuan/python-solvespace/commit/24cfed41705539f42df099bf45405b652b4a6778. They are not existed in the current version.

Wrapper source code:
+ `Cython/slvs.pxd`
+ `Cython/slvs.pyx`

Compile settings:
+ `Makefile`
+ `requirements.txt` (Python dependence)
+ `setup.py`
+ `.travis.yml`
+ `appveyor.yml`
+ `platform/patch.diff` (for Python compiler option update)

For development:
+ `__init__.py` (module expose for Python code)
+ `slvs.pyi` (stub file)
+ `test_slvs.py` (unit test)
+ `example/*.slvs` (cases in unit test)

The makefile is just used to call "`python setup.py build_ext -j0 --inplace`" and clean them up (non-essential). In the current test, Python-Solvespace can be coexist with Solvespace.

To merge them, there must be some work here:

+ [ ] Move above source code into `Cython` directory.
+ [ ] The `setup.py` file needs to adjust as https://packaging.python.org/tutorials/packaging-projects/.
+ [ ] Integrate CI tests.

Directory is as follows:

```bash
/solvespace
    /Cython
        /Python-Solvespace
            __init__.py
            slvs.pyi
            slvs.pxd
            slvs.pyx
        /platform
            patch.diff
        /tests
            __init__.py
            test_slvs.py
        setup.py
    ...
```

whitequark, 2019-05-24:
Looks reasonable.

KmolYuan, 2019-05-28:
There are two different ways to add this solver wrapper.

Let [solvespace] as submodule in [python-solvespace]; or integrate the wrapper into the branch (or maybe new branch) of [solvespace], then [python-solvespace] might be deprecated.

[solvespace]: https://github.com/solvespace/solvespace
[python-solvespace]: https://github.com/KmolYuan/python-solvespace

I think the former will be more convenience to publish the wrapper to PyPI with CI service. The submodule can follow `master` branch if Solvespace updated.

If the latter one, the wrapper packages still can released at a forked repository.

@whitequark Which way does this community prefer?

whitequark, 2019-05-28:
So, there's one thing to discuss first. SolveSpace uses the GPLv3+ license and has a CLA. Are you willing to relicense the wrapper as GPLv3+ and sign the CLA? If no, the bindings unfortunately have to be separate. If yes, then I propose putting the bindings in the same repository, so that they will always be tested with the most recent SolveSpace version, and distributed together.

KmolYuan, 2019-05-28:
OK, I will change the license of [python-solvespace] to GPLv3+.

I will fork this repository and do it at `python wrapper` branch as above plan.

[python-solvespace]: https://github.com/KmolYuan/python-solvespace

KmolYuan, 2019-06-06:
The [PyPI](https://pypi.org) releases require user account and password from CI serves. In common way, the user name and password stored in `TWINE_USERNAME` and `TWINE_PASSWORD` environment variables of Twine.

Travis CI (Ubuntu):

```yaml
matrix:
    include:

        - os: linux
          sudo: required
          python: "3.6"
          dist: xenial
          deploy: &deploy-options
              provider: pypi
              user: $TWINE_USERNAME
              password: $TWINE_PASSWORD
              skip_cleanup: true
              skip_existing: true
              on:
                  tags: true

        - os: linux
          sudo: required
          python: "3.7"
          dist: xenial
          deploy:
              <<: *deploy-options
```

Travis CI (Mac OS):

```yaml
after_success:
    - if [[ "$TRAVIS_OS_NAME" == "osx" ]]; then
        if [ -n "$TRAVIS_TAG" ]; then
            python3 -m pip install twine --user;
            python3 setup.py sdist bdist_wheel;
            python3 -m twine upload dist/*.whl --skip-existing;
        fi
        fi
```

AppVeyor (Windows):

```yaml
artifacts:
    - path: dist\*.whl

deploy_script:
    - IF "%APPVEYOR_REPO_TAG%"=="true" (pip install twine)
         ELSE (echo deployment has been skipped as environment variable has not matched)
    - IF "%APPVEYOR_REPO_TAG%"=="true" (twine upload dist\*.whl --skip-existing)
```

---

Here are some things to do:

+ Change path of package.
+ PyPI account of solvespace (and contributors).
+ When to release? (by tags?)

whitequark, 2019-06-11:
@KmolYuan Thanks for preparing this. I'll look into it closer shortly.

KmolYuan, 2019-09-25:
Hello! I have published the wrapper named as `python-solvespace` on [PyPI](https://pypi.org/project/python-solvespace/) (under GPLv3+).

```bash
pip install python-solvespace
```

The CI / CD configurations are modified with "matrix" mechanism ([Travis](https://travis-ci.org/KmolYuan/solvespace) / [AppVeyor](https://ci.appveyor.com/project/KmolYuan/solvespace)). And I keep the old deployments on `solvespace/solvespace`, PyPI deployments on `KmolYuan/solvespace`.

I hope you guys can check them out if I do something wrong. Then I will open a pull request.

+ [.travis.yml](https://github.com/KmolYuan/solvespace/blob/python/.travis.yml)
+ [appveyor.yml](https://github.com/KmolYuan/solvespace/blob/python/appveyor.yml)

The distributions are supported with Windows (`3.6`, `3.7`) and mac OS (`3.6`, `3.7`, `3.8-dev`), Linux user needs to build from source.

The package version `3.0.0` is depend on C++ kernel just like `PyQt`, the major and minor version are same, the micro version will changed with wrapper updating.

Symbian9, 2019-09-25:
> Hello! I have published the wrapper named as `python-solvespace` on [PyPI](https://pypi.org/project/python-solvespace/)

Awesome!

But check "Project description" on PyPI website - seems like its markup is broken! (does PyPI support Markdown markup?)

KmolYuan, 2019-09-25:
@Symbian9 Oh, that's a upstream issue that PyPI can't update description unless release a new version.
(I forgot to set `text/Markdown` first time.)

Maybe I can publish `3.0.0.post0` to fix this.

KmolYuan, 2019-09-26:
Oh! I made a big mistake with Linux sources.

The upper directories and files needs to move into `python_solvespace` folder, otherwise they will excluded by `sdist` (packing sources) command. I will fix them in a while.

igorkb, 2020-10-21:
I am sorry but it's not clear. @KmolYuan, will you sign the CLA and merge the wrapper?

KmolYuan, 2020-10-21:
@igorkb Yes, I have signed. Python wrapper is at a stable version `3.0.3.post1` currently.

I can open a new PR if there comes another release. (maybe after a long time maintenance)

https://github.com/solvespace/solvespace/pull/493

ruevs, 2020-10-28:
Hi @KmolYuan have you noticed that one more person seems to have started/done a Python wrapper here?
https://github.com/realthunder/solvespace/commits/python
It is quite out of date with master and I have not looked at the code (I do not know Python) but perhaps you and @realthunder could join forces. Or you could look at his for for useful pieces/ideas.

KmolYuan, 2020-10-28:
@ruevs Well, that wrapper is made by SWIG, which needs to write an interface file and compile it together.
This version uses CMake to complete the compilation process.

My early attempts did the same.
But because it is not easy to maintain the Python port, this solution was abandoned.
The advantage of Cython is that the syntax is similar to Python, and it is easier to expand Python's functions (such as docstring).

The tricky thing is Cython sources can not compile directly with CMake. Maybe something like [scikit-build](https://scikit-build.readthedocs.io)?

realthunder, 2020-10-28:
In order to support assembly, the `libslvs` API needs to be extended. You can reference my patch [here](https://github.com/realthunder/solvespace/commit/a2b33c81a82cb158db2fa8748bbd161816b23abc). Although it is based on an older version, it should still be relevant.

Some other useful patches, which may or may not be relevant any more,

* Allow raising exception instead of `abort` on invalid input, (https://github.com/realthunder/solvespace/commit/b7c390d6941a23937d219b421fb62ab99964b09e)
* Expr memory optimization, as the original implementation for Linux is very inefficient (https://github.com/realthunder/solvespace/commit/1a38896c9d6edc393d7d25bd234c06f5da2fdd09)

vespakoen, 2020-11-24:
@realthunder I think those changes your pointed out would be really good to have in SolveSpace!
Would you consider making seperate PR's for those?
If not, would you mind if I would bring them over?

Has anyone ever considered making a (C++) Class that contains the solver only?
I imagine that would reduce duplication when making multiple bindings, I got this feeling because `embind` and `boost.python` seems to be pretty simple when using a Class.

realthunder, 2020-12-04:
@vespakoen I don't think I have time for making a PR here. You are of course welcome to use my code.

Symbian9, 2021-09-18:
### Awesome News!

Thanks to @hlorus, now there is *"Geometry Sketcher"* add-on for *Blender* `2.92+` (written on top of @realthunder's [`py_slvs`](https://pypi.org/project/py-slvs/) Python module) for adding *SolveSpace 2D geometry constraints into Blender!*

- Blender Add-On: https://github.com/hlorus/geometry_sketcher
  - Demo Video: https://twitter.com/hlorus1/status/1438913270098939904

N.B. This add-on implementation finally closes (partially) my request to @KmolYuan:

- https://github.com/KmolYuan/python-solvespace/issues/2